FreeCAD/FreeCAD #12481: [Problem] When you choose a part of the figure and try to create a sketch on it, the lines, points, arcs, etc. tools do not appear.

Pato1911, 2024-02-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When you choose a part of the figure and try to create a sketch on it, the lines, points, arcs, etc. tools do not appear. so you can't work on a project because these tools are essential

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Mexico (es_MX)
[/code]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-02-17:
Have you changed the background color to white and not adjusted the colors in Sketcher?
https://wiki.freecad.org/Preferences_Editor#Colors
https://wiki.freecad.org/Sketcher_Preferences#Appearance (this tab is called "Colors" in V0.21)

maxwxyz, 2024-02-18:
@Pato1911 can you please describe how to reproduce this?

Pato1911, 2024-02-19:
@maxwxyz Simply, if in an already made figure I tried to make a sketch on one of its faces, it did not allow me to use the tools already mentioned.
Also when I start a new project, to start it I can't create any figure because of that problem

maxwxyz, 2024-02-20:
@Pato1911 sorry I do not understand how to reproduce this, could you provide a demo file and clear instructions?

Pato1911, 2024-02-21:
The first step I do is start a new file to create a body followed by a part and a body, but when I later create a sketch, and when I try to use, for example, the line tool to make a shape and then extrude it, the tool simply does not appear

Here is an example of a project to which I cannot add elements due to this problem:
[Eje.pdf](https://github.com/FreeCAD/FreeCAD/files/14353103/Eje.pdf)

maxwxyz, 2024-02-21:
@Pato1911 this is an image, could you provide the .FCStd file?
You have to create a sketch inside a body if you want to use it in PartDesign.

Syres916, 2024-02-22:
@Pato1911 the reason you are struggling is you are not aware that there is an additional step required between two bodies in two files and that is called the Sub Shape Binder https://wiki.freecad.org/PartDesign_SubShapeBinder . This acts like the glue between the two bodies if that translates into Spanish!! You can then proceed to sketch on the binder.

maxwxyz, 2024-03-09:
@Pato1911 please let us know if the SubShapeBinder works for you, otherwise I'll close this.

Pato1911, 2024-03-09:
@maxwxyz I'm afraid that step didn't work.

maxwxyz, 2024-03-10:
Can you record a video on that problem?

maxwxyz, 2024-06-13:
@Pato1911 I'll close this issue. If there is further information feel free to reopen.

closed 2024-06-13 by maxwxyz