FreeCAD/FreeCAD #12489: [Problem] Creating symmetric a Body in PartDesign is cumbersome or even impossible

NomAnor, 2024-02-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When designing a symmetric part it is often advantageous to only model the symmetric half and then use a pattern to create
the final shape. After that non-symmetric features can be added.

Using the existing pattern tools makes this problematic or even impossible because they only work with additive/subtractive features and then pattern the tool shape (the shape that is added/subtracted). Having a long list of features it is cumbersome to select, esentially, all previous features. If some non additive/subtractive features are used (e.g. a pattern) that becomes then impossible.

Currently its possible to use the Part WB features to pattern the whole Body and then fuse it and use the result as a base feature
for a new Body. Then other features can be added. This results in lots of intermediary objects in the document and is not easy to use.

Let the pattern tools be able to pattern a whole shape not only a tool shape and the workflow becomes easier and more streamlined.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36005 (Git)
Build type: Unknown
Branch: main
Hash: c4daf256b2eb41fd75f8344d27890c14e1bcae08
Python 3.11.7, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.60.0
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.0.0
  * Assembly3 0.11.3
  * fasteners 0.4.21
```

### Subproject(s) affected?

PartDesign

### Anything else?

#### Related issues
There are multiple issues referencing the same problem but only for mirroring:
- #11947
- #8439 
- #5678
- #5705 

#### Example
@realthunder LinkMerge has a simmilar feature, a checkbox called "Transform sub-feature"
[image]

#### Solution
I have a working patch that adds a TransformMode property that allows switching between transforming the tool shapes and the normal shapes. I can create a PR after a bit of tweaking.

#### Points for discussion
In italic are my opinions.

- Implement new tools for this instead of changing the existing ones? _No, it will only confuse users what the differences between the tools are. This hurts discoverability_
- How to switch between the two modes?
   - A checkbox (LinkMerge)
   - A combobox (my TransformMode property)
   - An empty feature list (proposed in the above issues) _This is hidden to new users who don't already know about it_

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-19:
@NomAnor is this a dupplicate of: https://github.com/FreeCAD/FreeCAD/issues/5678 ?

NomAnor, 2024-02-19:
Kinda, its only about the mirror feature, but I think this should apply to all pattern features.

I thought about using an empty feature list to switch to this mode. For the implementation this might be fine. But I see a problem in the UI. For the user there would be three state:

1. Non-empty feature list. Feature tool shapes are transformed
2. Empty feature list. User removed all features and wants to add new ones.
3. Empty feature list. User wants to transform the shape of the base feature.

2 and 3 are indistinguishable in the task panel, but they should behave differently.
2 Should do nothing until new features are added to the list. I think this is how it currently works.
3 should show the result of the transformation of the bease feature shape.
A user that removes all features might be confused when the base features is transformed.

There could be a label stating the current mode, but user rarely read those.
Better would be a pair of radio buttons, but that would require a new mode property to be able to distinguish 2 and 3.

NomAnor, 2024-02-24:
I created a draft PR for a solution. It is based on other PR so the diff looks a bit wild until they are merged.

This will transform the base feature's shape if no features are selected. What is missing is better discoverability in the UI.

PR #12589 (2024-06-03): Add a new mode to PD patterns that will transform the base feature's shape instead of the tool shapes
This allows applying the patterns to the whole base feature's shape. 
This makes it easier to design symmetric Bodies, especially if the feature list contains non-additive/subtractive features.

As mentioned in the issue, I think this requires some help on the UI side to make this new mode discoverable for the user.
Does someone have an idea for this?

This would resolve the following issues:
- fixes #12489
- fixes #11947
- fixes #8439
- fixes #5678

This is based on my refactoring PRs, so I w [... cut]
closed 2024-06-03 by chennes