FreeCAD/FreeCAD #12490: [Problem] Assembly workbench Distance joint not working

macdroid53, 2024-02-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When attempting to set a zero distance joint between vertexes it fails to adjust the location.
[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36012 (Git)
Build type: None
Branch: main
Hash: 62c8469bded2f39099c7b067889052818b659fe9
Python 3.11.6, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * Curves 0.6.22
  * CurvedShapes 1.0.5
  * DynamicData 2.60.0
  * OpenDark 2023.12.17
  * lattice2 1.0.0
  * sheetmetal 0.4.0
  * Manipulator 1.5.7
  * kicadStepUpMod 10.22.9
  * QuickMeasure 2022.10.28
  * fasteners 0.5.0
  * Help 1.0.3
  * ProDarkThemePreferencePack 1.0.0
  * freecad.gears 1.0.0
  * ExtremeProDark 2.9.0
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-19:
@PaddleStroke also concerning the distance joint

PaddleStroke, 2024-02-22:
Can you please attach the files so I can test?
It seems the JCS is not positioning correctly.

macdroid53, 2024-02-24:
- Create a new document
- Create a Part WB solid (Sketch and Extrude)
- Save
- Create a new document.
- Create a Part Design Body, Sketch, and Pad
- Save
- Create a new document
- Create an Assembly
- Save
- Click Insert link
- Tick "Insert as Part"
- Insert the the Body and Extrude from above files
- Attempt to add a Distance (or any) constraint

With source 36120, produces the same results as shown in movie earlier in this issue thread.

macdroid53, 2024-03-01:
> @PaddleStroke also concerning the distance joint

I gave instructions above on how to test.

macdroid53, 2024-03-04:
Is this stalled?

PaddleStroke, 2024-03-05:
@macdroid53 do not worry I will fix it. I have just not come to it yet.

PaddleStroke, 2024-03-11:
I cannot produce this error (in [my current branch](https://github.com/FreeCAD/FreeCAD/pull/12406) at least).
[image]

First is that I killed the 'insert as part' option that made little sense. But even recreating the same as with insert as part (basically it created a part and put the link in the part) I cannot reproduce the problem.

Now I'm not sure if this bug is fixed in my branch or if I have not reproduced the exact same file.

Please make sure to attach the file when you report a bug so that it removes uncertainty regarding it's reproduction.

Any chance you can try to reproduce this bug in my dev branch?
Thanks

macdroid53, 2024-03-11:
I have no file that shows the issue. Once the file has been saved the behavior is different. I am happy to provide a file when it shows the issue...in this case there is none.

This image shows what I get after "Insert as part" is used and I attempt to make a constraint.

Note the vertex selected and the axis that is associated. This does not happen is the "Insert as part" is not used.

It does not happen if the file is closed and opened again (I just noticed this) (but, after reopening the file, attempting to create a constraint crashes FreeCAD.)

[image]

I'm happy to build your branch, unfortunately, I know how to build main, I know how to build a PR...but, I don't know how to get your dev branch source to build. Sorry.

maxwxyz, 2024-03-11:
https://github.com/FreeCAD/FreeCAD/pull/12406 currently does not fix distance of 2 vertices with offset 0.
Also lines do not work:
[image]

PaddleStroke, 2024-03-14:
@maxwxyz I have put a fix for distance 0 between 2 vertexes. Was there a stand alone issue for this problem?

maxwxyz, 2024-03-14:
@PaddleStroke this one here

PaddleStroke, 2024-03-15:
Ah right, I got confused because later @macdroid53 mentioned problem concerning all joints.
So anyway the initial problem of the issue is fixed. Distance of 0 between vertex is fixed.

macdroid53, 2024-03-17:
> Ah right, I got confused because later @macdroid53 mentioned problem concerning all joints. So anyway the initial problem of the issue is fixed. Distance of 0 between vertex is fixed.

I mentioned that because it didn't appear to be the joints but the way the Part container was added to the solid it "Insert as part" was active and the user moved the cursor before click to drop the solid.

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre