FreeCAD/FreeCAD #12491: [Problem] Adding a transformation to a MultiTransform will not work correctly in all cases

NomAnor, 2024-02-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I can douple click on a PartDesign feature and edit its parameters even if the Body is not active.

Clicking on _Add * transformation_ in the context menu of a MultiTransform when the Body is not active will not add a new feature.

When a different Body is active than the one of the MultiTransform, the added transformation will reference the wrong Body (in its ``_Body`` property) and the transformation will fail with the message:

> PartDesign::MultiTransform: Link(s) to object(s) 'Mirrored' go out of the allowed scope 'MultiTransform'. Instead, the linked object(s) reside within 'Body'.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36005 (Git)
Build type: Unknown
Branch: main
Hash: c4daf256b2eb41fd75f8344d27890c14e1bcae08
Python 3.11.7, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.60.0
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.0.0
  * Assembly3 0.11.3
  * fasteners 0.4.21
```

### Subproject(s) affected?

PartDesign

### Anything else?

The action handler function is aborted when there is no active Body. And the wrong body is obvisusly used when a different one is active.
https://github.com/FreeCAD/FreeCAD/blob/62c8469bded2f39099c7b067889052818b659fe9/src/Mod/PartDesign/Gui/TaskMultiTransformParameters.cpp#L289-L297

It is probably enough to replace the active body with the body of the MultiTransform.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-03-18:
Wouldn't it make sense to issue a message saying "you are trying to change an inactive body.  please make the body active and try again."

NomAnor, 2024-03-18:
There is no message if you edit a sketch or other feature of an inactive body. I think such a message would be annoying.

PR #12538 (2024-04-01): Allow adding subfeatures to MultiTransform if there is no active Body
Previously the active Body would be checked and used to add new subfeatures.
This would either do nothing, when no Body was active, causing confusion for users becaues than can edit other features without an active Body.
Or it would add the subfeature to the wrong Body if another Body than the one of the MultiTransform was active.

Now the Body of the MultiTransform is checked and used. This fixes #12491.
closed 2024-04-01 by chennes