FreeCAD/FreeCAD #12494: [Problem] PartDesign scale transformation does not work with multiple features

NomAnor, 2024-02-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A scale transformation in MultiTransform only works correctly on the first feature in the feature list.

[image]

The hexagonal hole is scaled correctly, but the round hole shifts because it is scaled based on the center of mass of the hexagonal hole.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36005 (Git)
Build type: Unknown
Branch: main
Hash: c4daf256b2eb41fd75f8344d27890c14e1bcae08
Python 3.11.7, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.60.0
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.0.0
  * Assembly3 0.11.3
  * fasteners 0.4.21
```

### Subproject(s) affected?

PartDesign

### Anything else?

#### Cause
The problem is expected, looking at the comment in the code:
https://github.com/FreeCAD/FreeCAD/blob/62c8469bded2f39099c7b067889052818b659fe9/src/Mod/PartDesign/App/FeatureScaled.h#L54-L63

This essentially results from the way the patterns are implemented. The ``getTransformations()`` function returns a list of transformations that are applied to the features:
```python
transformations = getTransformations(features)
for f in features:
  for t in transformations:
    tool_shape = f.transform(t)
    base_shape.fuse(tool_shape) # or cut
```
This requires two properties of the transformations:
1. The transformation is based on the LCS of the Body
2. The transformation is independent of the feature

Both are broken by the scale transformation:
1. The transformation is based on the center of mass
2. The center of mass is a property of the feature

The failure is mitigated for a single feature by providing the list of features to the ``getTransformations()`` function. But because the list of transformations is calcuated before the loop, this will only work for the first feature.

This also results in code duplication in MultiTransform and special handling for the scale transformation.

#### Solution
Instead of ``getTransformations()`` implement ``std::vector<TopoDS_Shape> applyTransformation(std::vector<TopoDS_Shape> const& shapes)``.

Then all transformation features can do what they need without the above requirements and the code should be cleaner.
Input is a list of shapes that should be transformed and the result is another list of shapes. The loop would be something like this:
```python
for f in features:
  tools = applyTransformation([f.shape])
  for t in tools:
    base_shape.fuse(t) # or cut
```
The use of ``std::vector<>`` is needed for MultiTransform:
```python
prevShapes = shapes
newShapes = []
for st in sub_transformations:
  newShapes = st.applyTransformation(prevShapes)
  prevShapes = newShapes
return newShapes
```
This is a rough sketch, there are probably some details I'm missing.

#### Change of behaviour
This could result in a small performance decrease because the transformations are not precalculated (and pre-multiplied by MultiTransform). I assume the scale transformation is rarely used, then this can be mitigated by providing the algorithm with the information if the transformations are independet of the features and then a code path with the same performance as currently can be used.

This could also break older files when they use the scale transformation with multiple features, but I would assume that is very
rare.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pieterhijma, 2024-03-18:
This is in response to my review [here](https://github.com/FreeCAD/FreeCAD/pull/12537#pullrequestreview-1943163017). 

I've tried the example and what happens above is actually what I would expect. For the multi-transform two features are chosen, the hexagon and the circle. I first apply a linear transform and then a scale. I expect the scale to apply to all of what the linear transform captures and in my opinion that is also the distance between the hexagon and circle.

The fact that the holes overlap with the neighboring hexagon is not exclusive to this: it would be the same outcome if you have only the hexagon and you scale it without enough space for all the occurrences.

So, I understand that you want the distance between the hexagon and circle to be constant, but in my humble opinion you should accomplish this with two multi-transforms, one on the hexagon, and one on the hole, connecting the length and occurrences with variables.

So, let's turn it around. Suppose we would accept #12537, could we still create a scale transformation where the distance between the two features is scaled as well? I think it would be very difficult to accomplish. So, in my opinion the current multi-transform is what I would prefer in terms of semantics:

[image]

With #12537, it becomes:

[image]

I can see use for this, but I don't think it should be done with a single multi-transform then, but with two. How could I scale the distance between the hole and hexagon as well with the version from the PR?

Perhaps I'm missing something. In that case, could you explain @NomAnor?

NomAnor, 2024-03-18:
I understand what you mean. It might still be possible to use the shape implementation by changing the nesting of the for loops, but I have to think about it.

When initally looking through the code, I saw these comments:
https://github.com/FreeCAD/FreeCAD/blob/254d8ac7f0c2def5d7b6f8fbfdc75cea33ddd885/src/Mod/PartDesign/App/FeatureScaled.cpp#L67-L68

https://github.com/FreeCAD/FreeCAD/blob/254d8ac7f0c2def5d7b6f8fbfdc75cea33ddd885/src/Mod/PartDesign/App/FeatureScaled.h#L58-L59

These made me believe that the intended behaviour was to scale all features based on their own center of mass and not on the center of mass of the first feature. And that this was just not implemented at the time because it is not possible with the transform approach.

I don't have a use case for the scale feature, so I don't know what would be the more useful behaviour. Does anyone have an example where the scale feature is used or is useful?

AjinkyaDahale, 2024-05-21:
A few things I could add:
1. When scaling, an additional point on the base of the patterned feature (or maybe the patterned feature itself if you consider 3D scaling) should be given. This would be the point that is patterned without scaling, and all other points on the feature are scaled w.r.t. distance from this.
2. The entire feature should scale together, i.e. as per the first first image in (this comment)[https://github.com/FreeCAD/FreeCAD/issues/12494#issuecomment-2004155775]. For the second option it should be another feature. Thus, if the hexagon and circle are from the same sketch, it should be like case 1, if different case 2.

maxwxyz, 2024-11-10:
@FreeCAD/design-working-group @FreeCAD/cad-advisory-group any input on how it should be resolved?

pierreporte, 2024-11-10:
The main problem is that there can be two ways to think about a transformation of multiple features:

1. the current one, considering the selected features as a set, to which is applied the transformation ;
2. applying the transformation to each of the selected features separately.

If 1 is desirable, then is is valid. There could be a better UX but there wouldn’t be an actual problem. If 2 is more desirable, it could be a good idea to allow to select multiple features. If both are wanted, providing a set of radio buttons to chose between the two behavior would be a good idea.

I would prefer to stick to the current behavior, but improving UI/UX so that it is more obvious. For example, the barycenter of features should be displayed, and the user should be able to manually select it. This way, users will see that the result may not correspond to what they want so they can take action, and they would have better control. They would still be able to make multiple transformations for point 2.

pieterhijma, 2024-11-11:
For what it's worth, I would recommend to close PR https://github.com/FreeCAD/FreeCAD/pull/12537 and this issue. In my opinion, merging the PR would make it difficult to scale two features and the distance between it, whereas with the current way transformations work, it is possible. I definitely that the UX can be improved, but I would create a new issue for that.

NomAnor, 2024-11-20:
I've closed the PR. Looking at the comments, the current behaviour has it's use. If you want to close the issue it's ok with me.

luzpaz, 2024-11-20:
>  I definitely that the UX can be improved, but I would create a new issue for that.

@NomAnor do you want to go ahead and do that?

closed 2024-11-20 by luzpaz