FreeCAD/FreeCAD #12495: [Problem] Sketcher_Translate: "Spinbox doesn't exist" warning

Roy-043, 2024-02-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Sketcher_Translate tool displays a "Spinbox doesn't exist warning".

1. Open a sketch
2. Create a circle.
3. Select the circle.
4. Call Sketcher_Translate.
5. Pick a base point.
6. Move the cursor.
7. Result: warning.

Text of the warning:
```
20:57:47  EditableDatumLabel::setSpinboxValue: Spinbox doesn't exist
```

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-19:
Log occurs on my side right after selecting the tool

Roy-043, 2024-02-19:
The Sketcher_CreateRegularPolygon tools have the same issue.

luzpaz, 2024-03-15:
An proposed solution ?

wwmayer, 2024-03-16:
The problem is caused by SketcherToolDefaultWidget::eventFilter

When moving the cursor then setModeOnViewParameters() is called that calls EditableDatumLabel::startEdit() to create the spin box. This also calls EditableDatumLabel::setFocusToSpinbox() to set the focus on the created spin box.

Because a widget loses focus the installed event filter EditableDatumLabel::setFocusToSpinbox() jumps in and reacts on the FocusOut event by emitting the signalParameterFocusOut(). This signals is handled by DrawSketchDefaultWidgetController that calls EditableDatumLabel::setFocusToSpinbox() for all its EditableDatumLabel instancs.

Because at this point only the first EditableDatumLabel has a spin box but the others not yet because EditableDatumLabel::startEdit() wasn't called for them.

So, the problem is that SketcherToolDefaultWidget::eventFilter is called too early.

wwmayer, 2024-03-16:
A solution to the problem is that SketcherToolDefaultWidget installs itself as event filter to all its EditableDatumLabel not inside the constructor but after it has created all spin boxes.

PaddleStroke, 2024-03-17:
I think this bug is fixed by https://github.com/FreeCAD/FreeCAD/pull/12641

closed 2024-03-22 by chennes