FreeCAD/FreeCAD #12496: [Problem] Sketcher_Translate: focus can jump between Copies and Rows inputs

Roy-043, 2024-02-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After using the Tab key to change the focus to the Rows input, the focus can start jumping between the Copies and Rows inputs when the cursor moves.

1. Open a sketch
2. Create a circle.
3. Select the circle.
4. Call Sketcher_Translate.
5. Pick a base point.
6. Use Tab to change the focus to the Rows input.
7. Move the cursor.
8. Focus jumps repeatedly between the Copies and Rows inputs.

The preview as you drag the mouse seems a bit sluggish as well BTW.

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-28:
Can you check again? It doesn't seem to happen in the newest weekly build but it would be good if you could confirm.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

Roy-043, 2024-06-28:
This has indeed been fixed.

closed 2024-06-28 by Roy-043