FreeCAD/FreeCAD #12500: [Global Problem = request functions] Repeat last action

kaiwas, 2024-02-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

At the moment, there is no such important action in the program as “repeat the last action”. Because of this, you need to enter values every time you use tools (move/copy/rotate/chamfer and everywhere).
Is it possible to implement at least some kind of checkbox that would save the entered data for reuse?
But in fact, repeating a useful action would be useful absolutely everywhere. This greatly speeds up repetitive tasks.

As an example, I am attaching a video to describe the problem in freecad and how it is implemented in blender/inkscape/coredraw (and in other programs).
[image]

https://github.com/FreeCAD/FreeCAD/assets/32273833/52d15b28-42e5-43aa-acf2-750a64281bc4

https://github.com/FreeCAD/FreeCAD/assets/32273833/3dcc09bb-edec-4374-84e3-08300ae04eb7

### Full version info

```shell
Any version of freecad.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-04-06:
I personally do not see much in the scenario where you enter the same value again and again for Draft commands. In such a workflow you should use arrays instead.

You need to retest this with the current dev version. Copy mode is (re)stored properly in the dev version (and this was already the case in February).

kaiwas, 2024-04-07:
Despite the fact that there are similarities between the topics, these are TWO DIFFERENT PROBLEMS. They solve different problems (although there are overlaps).
In the case of "repeat last action CTRL+R" anything can be used (change color, delete or something). In this case, when performing repeated steps, there is no way to change those steps because the action is always the same.
In the case of working with fields (closed topic), we simply quickly perform a certain operation when working with fields and have the opportunity to change the parameters if necessary. For example, a pitch of 1000 mm. change to 2000 mm.. after the 5th copy.

If necessary, I can record an explanatory video that will demonstrate the difference between the two topics.

https://github.com/FreeCAD/FreeCAD/assets/32273833/1b6c9bb4-bf72-483a-b198-71dfc806743d

kaiwas, 2024-04-07:
> In such a workflow you should use arrays instead.

An array does not always meet the needs of quick copying, because it creates duplicate elements without the possibility of their subsequent editing separately from each other.
If there was a button to “destroy” the array, this would partially solve this problem.

> You need to retest this with the current dev version. Copy mode is (re)stored properly in the dev version (and this was already the case in February).

Thank you! Checkboxes do not change their meaning. This is good. (did not know). 
But the input fields are still reset to zero. Just think about it for a second - a value of zero is a completely useless value. In essence, this value is equal to the absence of action. It's as if we are starting the operation all over again every time.

maxwxyz, 2024-04-30:
this issue needs more discussion.