FreeCAD/FreeCAD #12507: [Problem] Sketcher_CreateRegularPolygon OVP: for center Y input has the 1st focus

Roy-043, 2024-02-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using On-View-Parameters to define the center in any of Sketcher_CreateRegularPolygon tools, the Y input box has the initial focus. You have to use Tab twice to get to the X input box.

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-19:
That would be an improvement to be consistent with all other tools.

Roy-043, 2024-02-23:
@PaddleStroke This is not a big issue obviously.  But IMHO it is something that should be fixed before the next release.

Roy-043, 2024-06-28:
This has meanwhile been fixed. Thanks.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
```

closed 2024-06-28 by Roy-043