FreeCAD/FreeCAD #12510: [Problem] Sketcher V0.22 Regression: horizontal/vertical constraint icon attached to cursor does not update properly

Roy-043, 2024-02-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As you pick the 2nd point of a line (for example) a horizontal constraint icon is attached to the cursor if the current point has the same Y coord as the first point. This icon does not update correctly in V0.22.

1. Click the 1st point of the line in the +X, +Y quadrant.
2. Move the cursor to the right: the horizontal constraint icon is attached to the cursor.
3. Move the cursor up: the horizontal constraint icon stays attached to the cursor. Only when the 2nd point is vertically above the 1st does the icon change to the vertical icon.

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-06:
@PaddleStroke is this also fixed with https://github.com/FreeCAD/FreeCAD/pull/15160 or https://github.com/FreeCAD/FreeCAD/pull/15161?

PaddleStroke, 2024-07-08:
No I don't think so.

PaddleStroke, 2024-07-08:
I don't understand how it was working before actually. because the tool handlers are doing : 

```
                if (seekAutoConstraint(sugConstraints[1], onSketchPos, onSketchPos - startPoint)) {
                    renderSuggestConstraintsCursor(sugConstraints[1]);
                    return;
                }
```

and seekAutoConstraint is false if no auto constraint detected. So it is clear that the icons are not updated if no auto-constraint detected.

PaddleStroke, 2024-07-08:
Looking at the end of the lineSet tool handler, which does not display this bug (it's a old-style DSH), I can find why this is so. The mouse move function is doing a `applyCursor();`

closed 2024-07-15 by chennes