FreeCAD/FreeCAD #12514: [Problem] Feature or Bug? Worksheet Toolbars not context switching

iAmVaeVictus, 2024-02-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Toolbars don't context switch when changing tiled windows. 
1. Open spreadsheet
2. Open Sketch
3. Tile windows
4. Click spreadsheet - tool bars and worksheet selection in dropdown do not change

### Full version info

```shell
OS: macOS 13.6.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

None

### Anything else?

Not clear if this is a feature but it seems to slow down my work as I have to keep changing to spreadsheet tools when working in the spreadsheet. Seems like context switching to the active windows would make sense, then allowing us to change the context in the dropdown if needed. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-20:
Not sure if this is a desired behavior. Currently no toolbars switch if you change documents.

maxwxyz, 2024-02-20:
@obelisk79 is this a wanted feature or not in line with UX?

obelisk79, 2024-02-20:
This really highlights a limitation of the workbench paradigm. I'm not entirely sure initially what the behavior should be. I'll have to engage with the rest of the DWG and have a conversation about it.

maxwxyz, 2024-06-13:
I'll close this issue for now as it works as intended and more objects now enable a WB / toolbar switch by double clicking on the tree icon (e.g. Spreadsheet, TechDraw Page, PartDesign Body, FEM,...)

closed 2024-06-13 by maxwxyz