FreeCAD/FreeCAD #12518: [Problem] The entered data does not fit into the formula editor

kaiwas, 2024-02-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The entered data does not fit into the formula editor (fx).
Example in video file.
I encountered this inconvenience a couple of times. The formula construction was long (different conditions) and its length did not fit into the input field.
I suggest replacing the text input line with a text field.
I think it's pretty easy to implement (I hope).

https://github.com/FreeCAD/FreeCAD/assets/32273833/179fa8b0-8d69-4911-bded-dd58108fb667

### Full version info

```shell
any version FreeCAD
```

### Subproject(s) affected?

Expressions

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jwueller, 2024-05-07:
You can assign this and #12591 to me, as I'll likely touch this in the PR for my proposal to improve the expression workflow overall.

luzpaz, 2024-08-17:
Is there a PR yet ?

jwueller, 2024-08-17:
I've been extremely swamped recently, so I didn't get it done yet. I'm still working on it though.

xtemp09, 2025-02-19:
@jwueller, is there any update on this topic?

B0cho, 2025-06-20:
If no one is working on that, I would implement fix for this as I'm working on other issue with Expression Editor. You can assign me.