FreeCAD/FreeCAD #12529: worst distortion on a simple mesh

Cyril4freecad, 2024-02-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have found something amazing.... Here it is.....
Make a cylinder RADIUS 10mm HEIGH 30mm with PART menu, go to MESH menu, click on A (analysis), select the cylinder, click on GMESH and select APPLY (dimension from shape, order 2nd). There is a warning message " Warning : Volume mesh: worst distortion = -0.00181706 (avg = 0.993714, 1 elements with jac" 
Now proceed exactly the same but taking 20mm or 40mm instead of 30mm for the cylinder length, and everything is normal, without warning. The trouble is for a length of 30mm. Amazing?!! I hope it will help you. 

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-21:
@Cyril4freecad can you try it in the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
and please also change the title to be clear and precise.
@FEA-eng FYI

FEA-eng, 2024-02-21:
I can't confirm it (tried in both 0.21.2 and newest 0.22 weekly build). Are you sure those are the right dimensions - radius 10 mm and height 30 mm ?

Cyril4freecad, 2024-02-21:
Hi, The radius is 2 mm instead of 10mm. Sorry for the mistake. 
Please find attached a clear mesh result with R= 2mm and H=20mm, and the mesh with a warning message with R=2 and H=30mm for the cylinder.  I have used the Gmesh. 
[image]
[image]

FEA-eng, 2024-02-21:
Ok, I can confirm it now. Both in 0.21.2 and 0.22 dev. Interestingly, it doesn't happen e.g. with H=40, as one could expect since the cylinder becomes more slender and thus more difficult to mesh properly with tetrahedral elements (but Gmsh has its quirks so I'm not that surprised).

However, I don't think there's anything we could fix here. Maybe the meshes should be more refined by default but it may not be possible to change if it depends on the defaults used by Gmsh algorithms. And even if FreeCAD can control the way this automatic mesh size selection works, it's recommended to use non-default settings and input the element size depending on the size of the smallest relevant geometric features. Here, you can get rid of warnings by entering even 1 as max element size.

Cyril4freecad, 2024-02-21:
I understand you can't do anything, as Gmesh is an independent "box" from Freecad, apart from forwarding the issue. Anyway, thank you for your time. Free cad is a good multi-tool.   :-)

FEA-eng, 2024-02-21:
Issues with Gmsh itself can be reported here: https://gitlab.onelab.info/gmsh/gmsh/-/issues but they have nothing to do with FreeCAD so I'm not sure if it makes sense to bother them with this. Unless you formulate it as a suggestion to consider changing the default mesh size determination.

Cyril4freecad, 2024-02-21:
OK noted, thanks.

FEA-eng, 2024-02-27:
@maxwxyz Can you close this as not planned?

closed 2024-02-27 by maxwxyz