FreeCAD/FreeCAD #12531: [Problem] VarSets cannot be exposed

pieterhijma, 2024-02-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This problem follows up on #12135. The overall problem is discussed in #12120. PR #12135 introduced a VarSet but it does not support the functionality described in #12120 yet. For convenience, the text below lists the topics that this specific issue addresses.

### Application Geometry Interface

The main workflow for VarSets is adding them to an `App::Part` object with the idea that the design of the part makes use of the properties in the VarSets.  However, in this case where the VarSet has an `App::Part` as parent, the VarSet becomes special. In that case, the VarSet will automatically obtain a boolean property `Exposed` in group `Base` that is initially false.  Users can simply continue to use the variables defined in the VarSet in their design. If they toggle `Exposed` to true, the parent `App::Part` will automatically obtain an `XLink` property with the same name as the VarSet that points to the exposed VarSet.  This `XLink` property will be `CopyOnChange` allowing users to change the link to the exposed VarSet to an equivalent VarSet outside of the `App::Part`.  Expressions used in `App::Part` that refer to the exposed VarSet will automatically be redirected to the equivalent VarSet.

This functionality allows Parts to have something similar to an API, let's call it an **Application Geometry Interface**. The designer of geometry expresses the parameters in terms of variables in VarSets. For example, suppose a designer creates a box with three parameters: `Width`, `Height`, and `Depth`. The designer creates an `App::Part`, with a VarSet `Parameters` and a PartDesign Body. The VarSet contains the three parameters with default values and the boolean `Exposed`.  The PartDesign body has been designed in terms of these three parameters.

By setting `Exposed` to true, the designer of the box communicates to users of the box that the design can be varied by means of `Width`, `Height`, and `Depth`.  The part will automatically obtain an `XLink` property with the same name as the VarSet, in this case `Parameters`.

Suppose there is a user of the box that wants to change the default dimensions of the box. The user can copy the VarSet within the part to a VarSet outside of the design, change the values of the three dimensions in this new VarSet, and change the `XLink` property `Parameters` of the Part to point to this newly created VarSet.  The geometry will change accordingly.

This solution gives designers and users of designs a well-defined interface with which the geometry can be changed from outside of the Part.  This provides FreeCAD and its design much improved modularity because parameters for geometry can be "injected" from outside into geometry via the interface that the designer defined for the geometry.

For example, consider an assembly of wooden parts created from the same sheet of wood resulting in subcomponents with the same thickness of wood. The assembly could have a VarSet that defines parameters that are shared among its subcomponents such as the thickness of the wood.  Then for each piece, you could define a VarSet with that thickness of wood, ensuring that the user has only one place where the thickness of wood needs to be changed, applying a new value for the wooden thickness to all parts of the assembly.

In conclusion, having a core property container allows us to define an interface between a design and users of the design, typically an assembly.

### Variant Links

With the above functionality, it is possible to create a link to the part, and change the Part property link to point to a different equivalent VarSet, essentially providing a convenient way to create variant links. 

Suppose the box above is assembled from a simple rectangular piece of wood with a common thickness.  The box has a long side, a short side, and a base.  With a simple rectangular sketch and a pad we create one of those sides with a VarSet that exposes the length, height, and thickness.

We can then define VarSets for each of these three variants and change the parameters accordingly. We then use one copy of the part 'side', make two links to this part (or alternatively two subshapebinders), change the property `Link Copy on Change` to `Enabled` and have the Parts point to their respective VarSets.

### Configuration management

By means of defining VarSets for a configuration, managing configurations becomes essentially changing a link from one configuration to another.  In the example above, the assembly uses three configurations defined by VarSets. What is especially powerful is that the configurations can come from outside.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35729 (Git)
Build type: Release
Branch: main
Hash: 8d88cc7e79e8e3bc9090ab8f1b244bc9a616e940
Python 3.11.6, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01
  * backup
  * BIM 2021.12.0
  * Assembly3 0.12.2
  * fasteners 0.5.0
  * Assembly4 0.50.6
  * osh-autodoc-workbench
  * DynamicData 2.60.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-21:
@pieterhijma can you split https://github.com/FreeCAD/FreeCAD/issues/12120 into separate issues?

maxwxyz, 2024-02-24:
I'm ok with that. If you have split #12120 into all individual issues, please don't forget to close it or pin it to a PR :)