FreeCAD/FreeCAD #12539: [Problem] Python API name clash: FreeCAD.Part is both a module and a class

snow-faerie, 2024-02-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

```python
>>> FreeCAD.Part
<class 'App.Part'>
>>> import FreeCAD.Part
>>> FreeCAD.Part
<module 'FreeCAD.Part' from '/tmp/.mount_FreeCA6HiINy/usr/lib/Part.so'>
```

The problem is actually more serious than a name shadowing inconvenience, as importing `FreeCAD.Part` will cause FreeCAD to misbehave and even crash. For instance, it will error out with messages like this when trying to open or operate on a sketch immediately after:

```
<Exception> SolverGeometryExtension.cpp(102): SolverGeometryExtension - notifyAttachment - Geometry not supported!!
```

### Full version info

```shell
OS: Fedora Linux 39 (Workstation Edition) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * fasteners 0.5.11
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-21:
Is this related to scripting or can this crash during "normal" behavior?

snow-faerie, 2024-02-21:
I found the way to import the Part module without corrupting FreeCAD, while reading the docs https://wiki.freecad.org/FreeCAD_Scripting_Basics#Creating_objects: `import Part`.

So, good news: it's possible to import the Part module without corrupting FreeCAD. The bad part is ofc that `import FreeCAD.Part` corrupting FreeCAD is a big risky footgun.

snow-faerie, 2024-02-21:
> Is this related to scripting or can this crash during "normal" behavior?

`import FreeCAD.Part` would have to be typed in the Python console, or be part of a macro that the user runs... but once that is done, FreeCAD while misbehave and crash while using the regular GUI, e.g. double clicking on a sketch.

luzpaz, 2024-03-20:
@wwmayer any ideas on how we users can avoid this ? TIA!

luzpaz, 2024-07-13:
@wwmayer any chance you can weigh-in ?

wwmayer, 2024-07-13:
The very simplest fix would be to remove this line: https://github.com/FreeCAD/FreeCAD/blob/9a51e2c6f1ae8b4a1919f6af0f90df6833ecf82f/src/App/Application.cpp#L333

This change should be fine as in Python code lines like this should be avoided anyway:
`issubclass(type(App.ActiveDocument.Part), FreeCAD.Part)`

PR #15409 (2024-07-15): Core: Fix Python API name clash: do not expose the App::Part type to …
…Python

Fixes #12539
commit 60090726 (2024-07-15): Core: Fix Python API name clash: do not expose the App::Part type to Python

Fixes #12539
closed 2024-07-15 by chennes