FreeCAD/FreeCAD #12550: [Problem] Assembly : Insert tool : 'Insert as part' option does not make sense

PaddleStroke, 2024-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Insert tool : 'Insert as part' option does not make sense and is apparantly confusing.
It needs to be removed.

Something more useful would be 'Show only parts'

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-03-06:
> Being also confused by this option (which was also enabled by default), and trying to understand best practices when creating an assembly:
>
> Bodies shouldn’t be standalone and shouldn’t be in assemblies
>
> Would you mind expanding on why bodies should not be standalone in an assembly? Thanks.

Because a body is the physical representation of a part, but a part is more than its physical representation. There can be several bodies as well. If bodies are inserted in assemblies, it is impossible to add more bodies to the part (say helicoils). Part attributes will be attached to the part container, not the body container.I think that it is important to understand that this wouldn’t be a limitation. It would be from a technical point of view but it makes data management easier without adding complexity to the user.

furgo16, 2024-03-06:
Thanks for the explanation. I do understand that Parts can contain multiple bodies, but I do not understand the advantages beyond that -unless you are using or planning to expand to multiple bodies parts.

For instance, in a very simple design of a box without a lid (bottom, side left, side right, front, back), where each component is a single body, I'm not certain what advantage it would bring to have each body contained within an extra Part each. If this wanders off topic, happy to take it to the forum instead.

pierreporte, 2024-03-06:
Two things:

1) It may work for simple cases but you can never know if it will need to be more complex. No later than today at my job, I had a very simple part in a very simple situation that ended up needing more than one body. It can happen at any time for anything. If you work right from the beginning, which shouldn’t be longer or harder, you wouldn’t even feel the need to insert standalone bodies in assemblies. I agree it’s currently not as easy today, if only because the standard part template doesn’t create a Part container as the root of the document.

2) Attributes of the part like name and part number (the most important one) need to be attached to the part, not to the body. If you insert a standalone body, it doesn’t bear that information. It’s just a solid. No name, no reference, possibly no material depending on the future implementation. Information management is highly important for CAD.

Basically, this often means nothing for hobbyists, similarly to the multi-root files issue (see #11742), but they are a true need for professionals. The limitation can mean artificial or unneeded but it is necessary to remember that CAD program mimick the physical world and that it makes no sense to allow impossible actions. At worst, users would shoot themselves in the foot.

qewer33, 2024-03-15:
> Bodies shouldn’t be standalone and shouldn’t be in assemblies. The standard part template should create a Part root object as well.

I personally disagree with this. Bodies are stand-alone solids and just like any other solid (eg. solids created via Part WB tools) they should be able to be used inside assemblies. Limiting assemblies to Std::Part containers does not make sense nor will it "fix" any apparent issue with the way the Assembly WB works.

Furthermore, even if such a change will be made, I've mentioned this in a related issue but Std::Part containers have critical bugs (transform issues, cyclic reference issues with multiple bodies etc.) that need to be addressed first. I remember there were existing Github issues for these problems but I can't seem to find them. Would appreciate if anyone could give it a quick search.

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre