FreeCAD/FreeCAD #12552: [Problem] PartDesign/Hole Counterbore diameter being blocked

csoete2, 2024-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have an older file that is being broken when opening the hole dialog since previous values don't seem to be valid anymore. It seems that one of the constraints is blocking the value (maybe the countersink angle that is not being used by counterbore?).

The gif is a sample project where I created a cube to demonstrate. When changing the counterbore diameter its getting blocked at 6.10mm no value below is accepted.

[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Germany (en_DE)
Installed mods: 
  * boltsfc 2022.11.5
  * sheetmetal 0.2.57
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[demo.zip](https://github.com/FreeCAD/FreeCAD/files/14376028/demo.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-02-22:
Backward compatibility is a tangent topic to discussion #11742.

jspiros, 2024-02-22:
I am still using the Debian 0.20.2+dfsg1-10 package of FreeCAD, and I regularly encounter this when creating new holes.

When the hole parameters panel first appears upon creating a new hole, I set the cut type to counterbore, and while the diameter field enables, it's stuck with 6.1mm as the minimum value. I can increase the value, but I can't type a value smaller than 6.1mm, and the "-" button is disabled when the value is 6.1mm.

The issue can be worked around, at least when I've encountered it, by creating the hole, setting other parameters, enabling counterbore, clicking "OK" to dismiss the hole parameters panel, then double clicking on the same hole feature to re-open the parameters panel, which will now allow values lower than 6.1mm for the counterbore diameter (assuming the other parameters shouldn't logically prevent it).

I haven't had any problems editing existing holes in saved files, but I don't have any files that were saved by any versions of FreeCAD older than 0.20.2. Even if editing a hole is the first thing I do after launching FreeCAD fresh and opening a saved project with existing sub-6.1mm counterbore diameter holes, the interface shows the correct sub-6.1mm values and doesn't prevent me from lowering the value even more if it's not already at the true minimum.

jspiros, 2025-06-18:
Checking with the latest weekly build, the hole task panel has been entirely redesigned, and does not seem to have this bug anymore. So, this issue can likely be closed,assuming the target is the latest dev version.