FreeCAD/FreeCAD #12555: [Problem] Spnav.h is missing from conda installation

domovinavi, 2024-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello,

i wanted to build the source code, which was configured with setup-environment.sh. 
I couldnt compile the code since spnav.h was missing in the include files.

I dowloaded the source code to spacenavd and compiled it. Afterwards i copy pasted it into the include and lib folders. Then it worked. 

Maybe inlcuding it in the environment.devenv.yml could automate this.

Thanks,
Domovinavi

### Full version info

```shell
OS: TUXEDO OS 2 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36100 (Git)
Build type: Release
Branch: main
Hash: 90ede977a399e8a446b2f27e17597f824825ea39
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * yaml-workspace 0.1.2
  * fasteners 0.4.72
  * A2plus 0.4.61
  * kicadStepUpMod 10.22.1
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-22:
@oursland can you help out here?

oursland, 2024-02-22:
`FREECAD_USE_3DCONNEXION` should default to `OFF`.  I am uncertain why this was enabled without the necessary pre-requisites having been installed.

domovinavi, 2024-02-23:
I had some issues with building it, as we talked in discord @oursland. 
Maybe some of the pre configurations i did with the
mkdir build
cd build
cmake ..
etc... 
left a flag somewhere set. 

Im going to try a clean build from start with the conda. Lets see if that is going to happen again.
Ill let you know.

greetings,
Domovinavi/domcivocato

domcivocato, 2024-02-23:
After a fresh freecad clone, with the conda setup.sh, the compilation worked without problems. So this was caused probably by build tries before. 

Thanks for your help tho,
Domcivocato