FreeCAD/FreeCAD #12556: [Problem] v0.22.0 Editing a contstraint by clicking on it in Sketch difficult

iAmVaeVictus, 2024-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Editing horizontal or vertical constraints on the Sketch by clicking the numbers is nearly impossible. Different behavior from v0.21. 
1. Add a constraint. 
2. Enter the value for it and hit return
3. Click the constraint to turn it green and then yellow.
4. Double click it to open the edit value box. 
5. Note that it does not open, but it rapidly cycles between status colors. 
6. The only way to get it to edit is to go to the constraints box on the left and clicking it from there. 

### Full version info

```shell
OS: macOS 13.6.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-22:
@iAmVaeVictus I cannot reproduce. You cannot edit values for a horizontal or vertical constraint. I've tried it with a horizontal distance but I do not know what you mean with yellow or green. Maybe you still have the dimension tool open when trying to click on existing dimension?
It works on my side without problems.

iAmVaeVictus, 2024-02-23:
@maxwxyz Hi Max, sorry for the unclear explanation. 

When you have a vertical constraint in a sketch, in my case parameterized to a value in a spreadsheet but I don't know if that matters, and I try to double click it to edit it, it won't open the edit dialog, sometimes. It's not consistent but I'm sitting here doing it now.  I'd say about 75% of the time it doesn't open. 

It's almost like the click radius of the numbers and the line that are used to create the constraint on the screen has been reduced. It's weird, it doesn't do this in the current released version. 

Here's how I recreate it: 
1. Open a sketch that has an existing vertical and horizontal constraint to be sure we are not in some other mode. 
2. Observe that the color of the constraint is orange (in my case, again, not sure if it matters). 
3. Mouse over to see the color change to yellow. 
4. Double click directly on the numbers for the measurement in the drawing. 
5. The colors of the constraint then quickly cycle from yellow-green-yellow. When I mouse away from it, it goes back to orange. 

As I said it is more often noticed on a vertical, but the horizontal constraint in my drawing is doing it now. This is true for any sketch I pull up. This sketch is just a simple box - 4 lines, 11 constraints. 1 horizontal and 1 vertical constraint. 

It is also true in different drawing files that I can open and repeat this on. 

I just opened another file and repeated 1-3 above. This time, as I've observed before, it works the first couple of times, but then when I switch back and forth to a vertical and a horizontal constraint, it goes back to the behavior I'm reporting - it color cycles from orange, to yellow (upon mouse over), double click, to green (selected) and then yellow (mouse over) in rapid succession. 

This is true also if I click the constraint 1x, turning it to green, then double click. Still doesn't work. 

Now here is what is significant that I just noticed: If I TRIPLE click, it opens. 

I'm on a Mac. Using a fully charged (100%) Magic Mouse. 

Uploading a screen recording. You can see my mouse clicks - I'm not slow clicking, nor have I changed my double click speed or any other settings between 0.21.2 and 0.22.0. 

[Screen Recording 2024-02-23 at 10.56.38 AM.mov.zip](https://github.com/FreeCAD/FreeCAD/files/14387966/Screen.Recording.2024-02-23.at.10.56.38.AM.mov.zip)

maxwxyz, 2024-02-24:
Ok you are talking about a horizontal length dimension, not the horizontal constraint :)
I've experienced the same issue

iAmVaeVictus, 2024-02-24:
> Ok you are talking about a horizontal length dimension, not the horizontal constraint :) I've experienced the same issue. Could you please edit the title to say dimension or dimensional constraint?

Yes, Sorry! Still learning. Changed title.

luzpaz, 2024-08-27:
added screencast to OT