FreeCAD/FreeCAD #12558: [Problem] Poor spreadsheet readability

Paddington2024, 2024-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Data entered into spreadsheet cells becomes very difficult to read when darker style sheets have been selected but all other items of text remain perfectly legible (Edit>Preferences>General>Style Sheet>Dark-Contrast)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * Assembly3 0.11.4
  * Assembly4 0.50.1
  * CADExchanger
  * cadquery_module
  * ExplodedAssembly
  * fasteners 0.4.56
  * FeedsAndSpeeds 0.5.0
  * freecad.gears 1.0.0
  * Help 1.0.3
  * Manipulator 1.5.7
  * Movie 2024.2.15
  * pcb
  * POV-Ray-Rendering
  * QuickMeasure 2022.10.28
  * Render 2024.1.23
  * sheetmetal 0.2.63
  * slic3r-tools
  * workfeature
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-23:
@MisterMakerNL

MisterMakerNL, 2024-02-23:
Some of these colors are set by the preference pack, this is why we moved away from using preferences and stylesheets. 
In 0.22 we have theme's the style-sheet can still be set on it's own in the theme tab, but is more for advanced users and shouldn't be necessary for anyone who don't want to tweak/play with it.
To prevent this from happening in older version is too only apply a preference pack and not a stylesheet. And reboot after applying.

maxwxyz, 2024-02-24:
I'll close this as a fix is implemented in 0.22 and a solution for older versions was proposed. Thanks!

closed 2024-02-24 by maxwxyz