FreeCAD/FreeCAD #12560: [Problem] The Path workbench name is confusing and unintuitive.

sliptonic, 2024-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Path workbench is for producing CNC toolpaths.  In most applicaitons this is called 'CAM' for Computer Aided Manufacturing.  
Typical users are confused.  They don't realize that FreeCAD has a CAM capability and they don't know how to use a 'Path' workbench.  

The term 'path' is also overloaded with other contextual meaning in the application.  We use paths to sweep objects.  We sometimes refer to file paths. 

### Full version info

```shell
Any
```

### Subproject(s) affected?

None

### Anything else?

When we first started building Path, we named it that because there was an older deprecated module called 'Cam' already.

That module has since been removed from the source so there's no namespace collision anymore.   

The best time to undertake a workbench renaming is prior to a 1.0 release.  Doing it later will only be more difficult. 

I'm not talking about changing much in the source where the name 'Path' is widely used.  Just the user-facing stuff.

This will, of course, impact documentation, and many 3rd party tutorials, books, and videos.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-02-23:
I'm glad that we share the same opinion. In Polish translation, after a voting, we decided to use CAM as the name for the Path workbench. I've even seen some complaints on a Polish CNC specialists forum that this workbench in FreeCAD has a strange name.

maxwxyz, 2024-02-23:
> > Will it be used for cnc paths only or is it planned to add other manufacturing processes (e.g. additive manufacturing)
> 
> This isn't intended to change functionality at all. Only name.

Yes sure, I was just asking if the intention in the future is to add more "CAM" functionalities or if this will be a "CNC" specific WB where other addons should focus on e.g. additive manufacturing, process planning,...
"CAM" for me is the overall name.

pierreporte, 2024-02-24:
@sliptonic Indeed it’s off-topic as I was bouncing off your remark. Let’s discuss this later in a dedicated issue but let me just answer quickly.

I don’t agree with you that this is on the design side. A part can be manufactured with various process and these processes will need different alteration. FDM and SLS are both 3D printing but one needs to stick to a bed and not the other. CAM could be a whole suite that allow you to prepare all kind of processes, providing necessary alteration features for all kind of manufacturing. Well, except maybe mold design.

burdickjp, 2024-02-24:
There are additive manufacturing (AM) processes which use CAM-ish toolpath strategies, such as directed energy deposition (DED) and additive friction stir deposition (AFSD). These are served in other CAD/CAM suites as CAM strategies specific to AM. NX has "multi-axis deposition" as a suite of CAM tools for AM. PowerMill and Hypermill both have CAM strategies for AM. MasterCAM has an extension called "A-Plus" by the company CAMufacturing (go Lancers).

more on topic, though, I think we can do better than "CAM". I believe "Manufacturing Workbench" would be sufficiently descriptive.

maxwxyz, 2024-02-24:
Sorry for opening up this manufacturing topic but I want to make sure to choose a name that really fits to the features or the roadmap of this WB.

sliptonic, 2024-02-24:
The correct name is 'Cam'.   That's what is widely used in other applications.  CAD/CAM is a well understood.  
As noted, the only reason we didn't use it before was the name collision with the older workbench.

pierreporte, 2024-02-24:
CAM can mean either machining or manufacturing. We can drop the computer aided part as it is obvious while in FreeCAD. Manufacturing is more general but CAM is more common. Maybe use CAM and refer to it exclusively as computer-aided manufacturing in the tooltips and documentation?

obelisk79, 2024-02-24:
CAM is the generally accepted industry term for this, my opinion is that changing the name is a really good step to disambiguating what functionality is present to (new/prospective/existing) users by significantly improving its discoverability. Particularly for people with previous CAM experience.

maxwxyz, 2024-02-24:
I'm also in favor of CAM (known like FEM). Only thought I had was that some might expect more features regarding overall manufacturing (as mentioned e.g AM).

JohnOCFII, 2024-02-26:
For the renaming -- Path --> CAM  makes perfect sense.  At some point there may be an **additional** workbench for additive manufacturing.  And if we go crazy, there might be a desire to re-organize workbenches into categories, and this whole category of stuff could be called "Manufacturing" at that time.  But for now -- Path --> CAM will help FreeCAD features be more discoverable (and understandable) to new users.

PR #12665 (2024-03-11): Rename the Path Workbench to CAM
Fixes #12560
closed 2024-03-11 by sliptonic