FreeCAD/FreeCAD #12563: Dashed construction lines in Sketch for better perceptibility

luisDias89, 2024-02-23:
### Original PR

https://github.com/FreeCAD/FreeCAD/pull/4246

### Description and Justification

The construction lines are not easily perceivable when multiple drawings are present simultaneously. I have experience with other CAD software and I notice that in all of them, construction lines have a different color (already implemented here) and are dashed. Additionally, there should be a small icon in the small menu on the left side indicating that it is a construction line.

A dashed pattern like this one:
[image]

More Info:
[image]

FEA-eng, 2024-02-23:
It's already implemented and available in weekly builds: #11996

maxwxyz, 2024-02-23:
@luisDias89 please try the lates builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
I'll close this as implemented.

closed 2024-02-23 by maxwxyz

luisDias89, 2024-02-23:
Thank you, you guys are fantastic. I've already given you two donations, this year I hope to give more. It's becoming fantastic