FreeCAD/FreeCAD #12565: Rebuild the Start WB

yorikvanhavre, 2024-02-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

## The problem

The Start workbench is currently entirely based on HTML. It displays a web page inside FreeCAD's MDI window, using FreeCAD's built-in Web Workbench. However, we wish to [remove the Web workbench](https://github.com/FreeCAD/FreeCAD/issues/12505) from FreeCAD. The Start workbench needs therefore to be fully rewritten using Qt widgets only, and no HTML.

## The new Start workbench

* Could be written either in C++ or Python (C++ preferred for speed as it's a default workbench)
* Show a **First time wizard** page
  * Already started on https://github.com/FreeCAD/FreeCAD/issues/11735
* Show **a main "projects"** page with:
  * Several icons to create a new project, with some scenarios/automatic defaults for Arch/BIM, PartDesign, etc
  * Recent files
  * Custom directories
* Show a **Help/documentation** page with:
  * Quick-start info for new users
  * Workbenches/addons info
  * Anything else? This section needs some good thinking and planning
* Show a **Blog** page with:
  * Contents fetched from https://blog.freecad.org (wordpress) and rendered without HTML (needs some parsing/conversion)
* A freshly installed FreeCAD cannot reach the internet without the user's consent. All internet operations need to check if internet connection has been allowed. The new Start WB needs a dialog to ask the user if they allow internet connection.
* Everything must be easily themable

## Expected Outcome

* Build a plan of your vision for a new Start workbench with mockups and visual ideas
* Work with the community, seek feedback, hear opinions, adjust your idea
* Design how the new Start Workbench will be coded
* Start coding!

## Project Properties

### Skills

The student should have reasonable coding skills in either C++ or Python. The student should also get familiar with:

* All the details and requirements on https://wiki.freecad.org/Google_Summer_of_Code_2024
* The [FreeCAD source code](https://github.com/FreeCAD/FreeCAD)
* How [workbenches work in FreeCAD](https://wiki.freecad.org/Workbench_creation)
* How the [current Start workbench](https://github.com/FreeCAD/FreeCAD/tree/main/src/Mod/Start) works
* Working with Git, GitHub and pull requests
* Be able to compile FreeCAD
* If working with Python, since there is a performance hit compared to C++ and possibly additional run-time dependencies, we ask the students to demonstrate they have the needed skills to do the job properly with the available modules, and demonstrate they understand the Python environment they'll have to work with in FreeCAD

### Difficulty

Medium

### Duration

175h

### Additional Information

* Main FreeCAD GSoC page: https://wiki.freecad.org/Google_Summer_of_Code_2024
* Potential mentor(s): [Yorik](https://www.forum.freecad.org/memberlist.php?mode=viewprofile&u=68)
* Organization website: https://freecad.org
* Communication channels: [GSoC section on the FreeCAD forum](https://forum.freecad.org/viewforum.php?f=46) and this issue

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-02-23:
> Show a main "projects" page with:
>
> * Several icons to create a new project, with some scenarios/automatic defaults for Arch/BIM, PartDesign, etc
> * Recent files
> * Custom directories

For this, you may be interested in https://github.com/FreeCAD/FreeCAD/discussions/11742#discussioncomment-7873460. It’s in a discussion but I can create an issue (and likely will). Depending on the strategy, you may want to work around this from the start.

yorikvanhavre, 2024-02-23:
> Depending on the strategy, you may want to work around this from the start.

Absolutely! A template system would be perfect to add here. One should however define how it would integrate in the Start WB. Maybe together with the project tab, have an additional templates tab? Maybe favorite templates are the ones that would appear on the project tab?

pierreporte, 2024-02-23:
Why not do like Solidworks? It has the welcome screen below, which is pretty close to FreeCAD. There are the standard built-in template, as with FreeCAD, and the custom templates are behind the “Advanced button”. I don’t know what it does but in older version, it showed the window visible in the linked discussion and copied below. Tabs are folder in the templates directory and the files are the contents of the folders. In FreeCAD, templates could be stored in `~/.local/share/FreeCAD/Templates`. I don’t know if sub-directories should be supported for FreeCAD though.

[image]

![image](https://private-user-images.githubusercontent.com/2029853/291015816-6b556385-62ac-4c1d-b816-67072743309f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2OTgxMTEsIm5iZiI6MTcwODY5NzgxMSwicGF0aCI6Ii8yMDI5ODUzLzI5MTAxNTgxNi02YjU1NjM4NS02MmFjLTRjMWQtYjgxNi02NzA3Mjc0MzMwOWYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjNUMTQxNjUxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2ZhZWYzMDMyNzYyZDI5OGYzNDg5Mjk0Njg4Yzg0NDdlYjI5Y2JmN2FiZTgyNzRmMjgwOTU3MzU2OTJiMTUxMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.gqFtg5j1o_brZKQorM0QnTlf8oBFZ5xQPVjdB4cT8uc)

pierreporte, 2024-02-23:
For favorites templates I propose this, on the basis of a Solidworks-like solution:

1. Have a checkbox somewhere to make the advanced templates (= custom) the default view.
2.  Allow to star custom templates so that they appear at the top of the list.

@obelisk79 What’s your opinion for how templates (built-in and custom) should be shown to the user?

obelisk79, 2024-02-23:
I like the general presentation of templates in the current development builds of  the Start WB. I think they are easy to understand and straightforward.

The ability to flag custom templates as 'favorites' for display seems like a reasonable approach. A simplified vs advanced view seems ok. 

I think we have to be cautious to not cram as much as we can into this new Start WB. It should be visually clean. Highlight the most used/important elements and avoid lots of things which are only situationally useful.

marcuspollio, 2024-02-25:
Hi there !

I'm with @obelisk79 here, we should not overload the Start WB, at least not from the _start_ of this new implementation, particularly as a GSOC project =D

Also not convinced displaying content from the Blog is that useful. Wouldn't a more general "News and communities" tab of simple links to the forum, blog, etc. with a nice visual composition be enough ?

yorikvanhavre, 2024-02-26:
Yes maybe. In any case, a large part of the work needed here is all the preparatory work I guess, building mockups and discussing them with the community. This should be very clear to any GSoC candidate, this project might be made of a lot of discussion and little actual coding

obelisk79, 2024-02-27:
Here is a mock layout created by @qewer33 originally posted on Discord
[image]

RohanMishra315, 2024-02-27:
Hey @yorikvanhavre I get the idea  what we have to do , we have to create a new workbench with totally written in QT widgets only no html. And now with the new workbench , we have to add the First Time Wizard Page , Main Projects Page, Help/Documentation Page, Blog Page . Well I think this require a lot of discussion to start with. I am working on creating some mockups design .

qewer33, 2024-02-27:
> Here is a mock layout created by @qewer33 originally posted on Discord.

I hava a couple more mock-ups in Figma that I initially created for my Start WB modernization PRs. I can share those here too if anyone is interested.

RohanMishra315, 2024-02-27:
Yeah @qewer33 pls share those too , it will help me to do some of the refinement bcz i am working on some and btw your mockups are great.

yorikvanhavre, 2024-02-27:
Those mockups are awesome! (Please share more @qewer33 !) I think that would help greatly people wanting to work on this.

Only little thing I would add on the setup page above: Some link or button or "what's this?" thing next to the nav style. Because a first time user might think: what the hell is that! :) But that might also be something to put in the help tab...

suryanshgargbpgc, 2024-02-28:
hi @yorikvanhavre, I want to work on this project this will be my first time building any major project using Qt, how should I start

yorikvanhavre, 2024-02-28:
@suryanshgargbpgc I would start with:

- Install and run FreeCAD to see how the start workbench works
- Familiarize yourself with Qt (an easy way is to follow the PySide tutorials on https://wiki.freecad.org/Power_users_hub )
- Study the start workbench code: https://github.com/FreeCAD/FreeCAD/tree/main/src/Mod/Start

qewer33, 2024-03-02:
Here's the Figma file and SVG exports for the mockups for anyone interested.

[FreeCAD Start Page Mockups.zip](https://github.com/FreeCAD/FreeCAD/files/14468332/FreeCAD.Start.Page.Mockups.zip)

RohanMishra315, 2024-03-02:
Hey @qewer33 thanks for these mockups . @yorikvanhavre I think the setup page and documents page are done and look good too although It can always be better. For the Help page , I thought of providing some startups tutorials for beginners or we can add some frequently asked questions type and For the Blog sections all the blogs will be integrated , @yorikvanhavre can you tell me what would be in activity section so I can think of and also what you think of above idea.

qewer33, 2024-03-02:
IMHO the activity section is unnecessary and can be removed. As for the help page, it would be great to offer some links to Youtube video tutorials in collaboration with FreeCAD tutorial creators (Mangojelly is the first name that comes to mind).

yorikvanhavre, 2024-03-04:
> IMHO the activity section is unnecessary and can be removed.

I agree partially... Yes, at the moment it serves little purpose, but I think we have an opportunity here. FreeCAD is not a "product". It's the hard work, the project, the baby of an entire community. When using a FOSS tool, you're not just "consuming" a product. You are taking part of a project. This has impact over you whole experience with FreeCAD.

I think, even if strictly unnecessary, we should keep something that works like a window that tells users "hey, this is FOSS, there is a lot more to see". Now maybe we can think of something better than what we currently do with it. Now that we have a good blog, maybe that's enough?

RohanMishra315, 2024-03-04:
@obelisk79 According to me (as a user) , I think the main startWB page is nice and effective as I first open freeCad I land on this page and as a curiosity I start looking around , so @yorikvanhavre has a point . And yeah , the page should be minimal and interactive.

marcuspollio, 2024-03-04:
About the "Activity" and "Blog" pages:

IMHO, I'm with @yorikvanhavre, there is an opportunity indeed : FreeCAD is not your typical proprietary CAD/BIM software...
This is a community-led project, FLOSS, striving for greater ethics for everyone (hobbyists and professionals alike), etc.

OK, you could say the same for other free software projects, but let's "ours" shine ! Isn't it one of the reasons for 1.0 ? XD

Now what (message to put forward) and how (integration in the Start WB), while being efficient and subtle ?

See below some other FLOSS "splashscreen/setup".
<details>
  <summary>Inkscape does it the tabbed way while exposing some (three) of its community artwork. The tab in the middle focuses on involving the users.</summary>

[image]

</details>

<details>
  <summary>Blender does it the more minimalist way with one artwork and a few links to its website.</summary>

[image]

</details>

What could be FreeCAD's way ?

obelisk79, 2024-03-05:
Don't get me wrong, I'm not saying there should be nothing. I suggested that perhaps the name "Activity" isn't ideal. Here is an example I created real quick that reduces number of options but still having access to the biggest social media communities, github and the blog. Less pages of the Start WB, less options but still highlights relevant stuff. This is just an example, so try not to hyper focus on the specific options I displayed here.

[image]

obelisk79, 2024-03-05:
Some other content ideas for this page: 'Community Favorite Youtube Channels' - ie highlight popular FreeCAD related youtube channels, bloggers, or a gallery view which can be fed selected images titled "Made with FreeCAD" I can draft some tweaks to this layout if requested.

yorikvanhavre, 2024-03-05:
I liked your mockup a lot @obelisk79 ! Yeah, that could work well!

JohnOCFII, 2024-03-05:
> Here is an example I created real quick that reduces number of options but still having access to the biggest social media communities, github and the blog. Less pages of the Start WB, less options but still highlights relevant stuff.

I like this -- and I like the use of the word, "Community."

mehulambastha, 2024-03-05:
Hi all, I just jumped onboard, and read the issue and all of your lovely and insightful comments!
Being a mechanical engineering undergrad myself, I've been on the "user" side of FreeCAD for quite a while and now I wish to get on the development side. I'm new to open-source, are there any pre-requisites to join this conversation, or can I jump right in with my tiny suggestions ?
Thanks :)

obelisk79, 2024-03-05:
@mehulambastha this is an open discussion. Share your thoughts.

maxwxyz, 2024-03-09:
I want to add point raised on the FreeCAD Day 2024 Complaint Session: 

**When you first open FreeCAD you end up on the Start workbench, when you open a new project the default Start WB is thewrong choice, no tools.**
The Start Workbench is incredibly unintuitive.
When you first open FreeCAD, you end up on the Start workbench, which you can't do anything useful with.
When you click on new project then there's nothing, no tools in the toolbar.
Right when you open your project, it doesn't. It doesn't allow you to select what workbenches start with, and the default choice is wrong.

**No tutorials for the examples**
There are examples that come and I can open example and see what someone has done, but it is not clear to me how I might find a description of how the example was made.
There are there are no tutorials or further documentation to go with the examples.

**To few example on the start page. Would like to see non workbench specific examples.**
By now they are separated by the workbench that generated that example file. Maybe simply adding more example of the various work entries with the different levels of complexity.
The the presentation of the examples is currently sorted by Workbench rather than by task, and it is not sorted by complexity of the example project.

RohanMishra315, 2024-03-12:
Hey @everyone I have design the help section could you take a look , sorry the design are very bad I'm not good with design but you'll get the idea
[image]
This is for the help section , I also like the idea of @obelisk79 "community one" . Please suggest

yorikvanhavre, 2024-03-29:
Hi all, just to notify that an effort is underway already to rebuild the start workbench in https://github.com/FreeCAD/FreeCAD/pull/13134

So it's best to consider this issue as no lounger a good candidate for GSoC. I'll remove the GSoC label

RohanMishra315, 2024-03-30:
That's sad to here @yorikvanhavre well I was preparing proposal on it. Even then, it's good that the effort is underway to work on it.

yorikvanhavre, 2024-03-30:
@RohanMishra315 have a look at the pull request, talk to @chennes , maybe there is more work needed that you can adapt into a GSoC project?

closed 2024-04-22 by sliptonic