FreeCAD/FreeCAD #12566: [Problem] Empty/Blank Section

frank-trampe, 2024-02-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This issue occurs for me in only one case, and that case seems similar to others in which it works. I cannot find an existing issue that describes it. A section view, placed to create a top view from a side view, fails to show a section line on its base view, and the section view is blank/empty. Attempts to delete and to recreate the section lead to the same result. Sections that cross the part work correctly. Sections placed at identical offsets from similar parts work for the 10 or so other parts added to drawings so far.

The file is attached below. The offending view is Section E - E on Page009. Note that the vertical offset is the same as for Section C - C, which correctly shows the top of a part of the same height as that to be depicted in Section E - E.

[Base_Framing_2_29_Package_1.zip](https://github.com/FreeCAD/FreeCAD/files/14385603/Base_Framing_2_29_Package_1.zip)

### Full version info

```shell
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) Snap 908
Build type: Release
Branch: tag: 0.21.2
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.15
  * dodo 1.0.0
  * BIM 2021.12.0
  * sheetmetal 0.4.0
```
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-23:
@frank-trampe could you try it with a current weekly build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

frank-trampe, 2024-02-24:
I tested with the weekly build (version information below), and the section remains blank, even when I try recreating it.

```
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36082 (Git) AppImage
Build type: Release
Branch: main
Hash: 7a025e4e60a607bdbdd1599627ce13bb8eeda071
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```

maxwxyz, 2024-02-24:
I can confirm this but cannot explain why this is the behavior

WandererFan, 2024-02-24:
Any chance you could reduce the file to Cut189 (Statue_Base_Frame_Middle_Ring_Beam_S_Complete_1), Page009, View035 and Section048 E-E?

It is too hard to find these items in the sample file.

frank-trampe, 2024-02-24:
The drawing is parametric, and there are a tremendous number of dependencies, so I probably cannot make a test case that slim. I am able to find a specific element by clicking into the tree view, pressing Control + F, and entering either the native name or the label in <<>>s. I am also able to select the item by selecting the view and then clicking the item name in the "Source" field.

WandererFan, 2024-02-25:
Your section origin does not intersect the base object.  The result of the section cut is empty.
11:07:30  DVS: SectionOrigin doesn't intersect part in SectionView
11:07:30  DVS::makeSectionCut - prism & input don't intersect - Section  - 
11:07:30  DVS::prepareShape - failed to build shape SectionView - Bnd_Box is void **

If you change your section origin to something that intersects the base view, like[0.00 -1,410.99 0.00] you'll get a valid section view.
[image]

closed 2024-02-25 by WandererFan

frank-trampe, 2024-02-25:
@WandererFan, it is generally common to use section views to provide top, bottom, and side views in a fashion that the reader can easily relate to the main view. You will note that this works for all of the other views on Page007, Page008, Page009, which makes me think that this is the correct way to make non-intersecting derivative views in FreeCAD. If it is not, would you point me towards instructions on how to make these views with the appropriate section lines? Even if there is a different way to make these, it seems unnecessarily complicated to have separate user actions and code paths for intersecting derivative views and non-intersecting derivative views, and I would advocate consolidating.

maxwxyz, 2024-02-25:
I get it that the section lines make it easier to see the direction but usually the different views are aligned depending on your projection method (first angle or third angle) https://werk24.io/knowledge-base-technical-drawings/projection-methods

frank-trampe, 2024-02-25:
It is difficult to visualize the significance of the "97.42° View" and the "127.4° View" (as occur on parts Statue_Base_Frame_Stair_Main_Cantilever_Beam_1 and Statue_Base_Frame_Stair_Auxiliary_Cantilever_Beam_1) without a section line. It also adds clutter and confusion when intersecting and non-intersecting views are drawn in sequence as for beam BB-2 on Page007. I have used FreeCAD for a long time and have used this functionality to do this specific thing for most of that time with no problems. I have trouble believing that the 99% of the time that I used this functionality and it worked was a result of error on my part and on the part of whoever wrote the code. It seems more likely to me that the 1% case where it does not work reflects a small oversight in the relevant code. I am happy to dive in if nobody else is interested, but, given that I have no experience with the FreeCAD codebase, it is likely to take at least an order of magnitude more time for me to find the problem than for somebody who knows the codebase to do so.

frank-trampe, 2024-02-25:
It is also worth noting that the cited warning (which I do not get) is of dubious applicability to the issue. It refers to <<Section A - A>> on <<Page>>, in which the section line actually does cross the part. I get similar warnings about many other views, and most of them render correctly.

```
16:48:33  DVS: SectionOrigin doesn't intersect part in SectionView021
16:48:33  DVS: SectionOrigin doesn't intersect part in SectionView023
16:48:33  DVS: SectionOrigin doesn't intersect part in SectionView025
16:48:33  DVS: SectionOrigin doesn't intersect part in SectionView027
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView028
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView029
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView031
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView032
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView033
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView034
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView035
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView036
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView037
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView038
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView041
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView042
16:48:34  DVS: SectionOrigin doesn't intersect part in SectionView043
16:48:35  DVS: SectionOrigin doesn't intersect part in SectionView044
16:48:35  DVS: SectionOrigin doesn't intersect part in SectionView045
16:48:35  DVS: SectionOrigin doesn't intersect part in SectionView046
16:48:35  DVS: SectionOrigin doesn't intersect part in SectionView047
16:48:35  DVS: SectionOrigin doesn't intersect part in SectionView048
```

frank-trampe, 2024-02-25:
Also, to be clear, I do get further warnings on the sections that fail to render as follows. My point is that the fact that the section origin is reported to be outside the part is not usually problematic.

```
16:48:38  DVS::makeSectionCut - prism & input don't intersect - Section E - E
16:48:44  DVS::prepareShape - failed to build shape SectionView048 - Bnd_Box is void **
...
17:02:25  DVS: SectionOrigin doesn't intersect part in SectionView050
17:02:26  DVS::makeSectionCut - prism & input don't intersect - Section G - G002
17:02:26  DVS::prepareShape - failed to build shape SectionView050 - Bnd_Box is void **
```