FreeCAD/FreeCAD #12568: [Problem] Selection of B-splines should be blocked for Sketcher_CreateFillet and Sketcher_CreatePointFillet

Roy-043, 2024-02-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It should not be possible to select B-splines for the two Sketcher fillet tools as they cannot handle splines. If you do select a B-spline the result is a not very user-friendly error message. Compare how circles are excluded from the selection. The same should be done for B-splines IMO.

Error message example:
```
15:31:58  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to fillet curves with ids : (33, 30) and points (81.9674, -71.9525, 0) & (71.3298, -61.9195, 0)
15:31:58  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').fillet(33,30,App.Vector(81.967421,-71.952536,0),App.Vector(71.329824,-61.919515,0),0.000000,True,True)
```

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-23:
Maybe it should be possible to add fillets to the start/end knots. You can manually create a tangent arc there.

AjinkyaDahale, 2024-02-28:
@Roy-043 could you replace "should be blocked" with "should not be allowed" to avoid confusion with the block constraint?

> Maybe it should be possible to add fillets to the start/end knots. You can manually create a tangent arc there.

It is possible, but he fillet tool does something more: it cuts part of the curves intersecting at the corner to add the circular arc without changing the remaining shape. That is however possible with some careful constraints and trimming.

[Screencast from 2024-02-28 18-29-45.webm](https://github.com/FreeCAD/FreeCAD/assets/4316933/85603828-e692-44ea-9479-a3659412dc80)