FreeCAD/FreeCAD #12570: [Problem] Strange artifact

vovodroid, 2024-02-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Strange artifact appears

[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English
Installed mods: 
  * Curves 0.6.13
  * ThreadProfile 1.87.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[mount.zip](https://github.com/FreeCAD/FreeCAD/files/14389324/mount.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-23:
@vovodroid please refine your issue as it is not clear what the problem is or how to reproduce it. You can also try to reproduce this in the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

vovodroid, 2024-02-23:
Ok, I updated the picture. Just open the project and you will see it.

vovodroid, 2024-02-23:
The same in latest (0.22.0dev.36082) dev version.

maxwxyz, 2024-02-24:
@vovodroid it seems like your Pad004 feature is already broken which may lead to this error. If you recompute, this is the result:
[image]
So it seems your Sketch007 is corrupted, try again and delete the Pad and the sketch and redraw this.
Your fillets had also some issues.

I've redone it here: 
[mount_corrected.zip](https://github.com/FreeCAD/FreeCAD/files/14392026/mount_corrected.zip)

I'll close this issue now as there is no bug identified.

closed 2024-02-24 by maxwxyz