FreeCAD/FreeCAD #12571: [Problem] Model|Tasks Window - last item covered by scroll bar

iAmVaeVictus, 2024-02-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When your Model | Tasks window on the LHS of the screen is not as wide as the text of the Tree item names, a scroll bar is generated at the bottom to scroll horizontally. When the tree is longer than the window and you must scroll to get to the end of it, the last item in the tree gets covered up (on a Mac anyway) with the scroll bar in the windows as you try to mouse over the item. The scroll bar actually "grows" taller when you mouse over it, completely obscuring the last item. 

It would be better if there was a little room at the end of the tree for that last item to be easily selectable without fussing with it. Screen grabs attached. 

[image]
[image]

### Full version info

```shell
OS: macOS 13.6.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-23:
Not sure if this is a bug but the default macOS behavior. @obelisk79 do you know this issue or is this changed with a theme?

obelisk79, 2024-02-23:
I have not experienced this before (but I don't use mac), it is possible that it could be mitigated using margins in the stylesheet, but that would be more of a hack than a proper fix.

iAmVaeVictus, 2024-03-06:
Adding more flavor to this with a quick video because it is frustrating, since I'm always dealing with the last element in the list, and I shouldn't have to collapse part of my tree just to get to the last item in the tree. I can click it, but it takes a frustrating amount of clicks to do it, or like I said, clicking to collapse part of the tree. 

https://github.com/FreeCAD/FreeCAD/assets/7082654/cf2dc828-c1b5-49fc-8ed7-773b0a455d04

obelisk79, 2024-03-07:
@iAmVaeVictus out of curiousity, would you be willing to install the OpenDark theme using the addon manager, apply it in preferences and report back if the situation is any different with it?

maxwxyz, 2024-03-17:
@iAmVaeVictus could you try this with OpenDark?
@MisterMakerNL FYI

MisterMakerNL, 2024-03-17:
I was like well if this is a mac issue surely there will be other programs with QT with similar issues.
But couldn't find any...
But I came across this topis about scrollbars in QT:
https://doc.qt.io/qt-6/qabstractscrollarea.html#details
We can maybe make it work for all os as in MAC:
https://doc.qt.io/qt-6/qt.html#ScrollBarPolicy-enum
The we can probably fix it without having to create empty placeholders which will probably get issues in other OS.

Issue here is that Qt::ScrollBarPolicy isn't the same on Mac compared to linux and windows.

JohnOCFII, 2024-03-18:
Another variable is that in macOS there is a choice of how to display scroll bars.  I've seen that in multi-platform toolset (such as QT) unless you choose the option to `Always` show scroll bars, odd things can happen.  Not sure if that is the case here, but I'd suggest that if it works with scroll bars set to `Always` show, any further change should be a low, low priority (and I say that as a Mac user).

  [image]

iAmVaeVictus, 2024-03-18:
My Appearances | Scroll Bar settings are "Automatically..." as above. 

I have installed the OpenDark theme and yes, that does solve the issue. It puts space below the last item, thus the last item is not obscured by the scroll bar. 

[image]

There is also a weird scroll issue if you swipe down on the mouse to get your tree to sc
roll to the bottom (but only when the tree is ALREADY at the bottom) and then move the mouse over to the actual drawing display window in a normal speed within 1-2 seconds of that scroll, then the drawing will zoom out as if I had done the mouse action in the display window to zoom out (Gesture Navigation style). That's very odd. Anyway, just mentioning it.

closed 2025-06-28 by github-actions[bot]