FreeCAD/FreeCAD #12575: [Problem] Draft Workbench modules are not loaded

FalcoGer, 2024-02-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening the draft workbench, I get errors about the draft modules not being able to load.

```text
00:09:32  <built-in function SoFieldContainer_getField> returned a result with an exception setError: Initializing one or more of the Draft modules failed, Draft will not work as expected.
00:09:32  Unknown command 'Draft_Fillet'
00:09:32  Unknown command 'Draft_Circle'
00:09:32  Unknown command 'Draft_Ellipse'
00:09:32  Unknown command 'Draft_Rectangle'
00:09:32  Unknown command 'Draft_Polygon'
00:09:32  Unknown command 'Draft_BSpline'
00:09:32  Unknown command 'Draft_Point'
00:09:32  Unknown command 'Draft_Facebinder'
00:09:32  Unknown command 'Draft_ShapeString'
00:09:32  Unknown command 'Draft_Text'
00:09:32  Unknown command 'Draft_Dimension'
00:09:32  Unknown command 'Draft_Label'
00:09:32  Unknown command 'Draft_AnnotationStyleEditor'
00:09:32  Unknown command 'Draft_Move'
00:09:32  Unknown command 'Draft_Rotate'
00:09:32  Unknown command 'Draft_Scale'
00:09:32  Unknown command 'Draft_Mirror'
00:09:32  Unknown command 'Draft_Offset'
00:09:32  Unknown command 'Draft_Trimex'
00:09:32  Unknown command 'Draft_Stretch'
00:09:32  Unknown command 'Draft_Clone'
00:09:32  Unknown command 'Draft_Edit'
00:09:32  Unknown command 'Draft_SubelementHighlight'
00:09:32  Unknown command 'Draft_Join'
00:09:32  Unknown command 'Draft_Split'
00:09:32  Unknown command 'Draft_Upgrade'
00:09:32  Unknown command 'Draft_Downgrade'
00:09:32  Unknown command 'Draft_WireToBSpline'
00:09:32  Unknown command 'Draft_Draft2Sketch'
00:09:32  Unknown command 'Draft_Slope'
00:09:32  Unknown command 'Draft_FlipDimension'
00:09:32  Unknown command 'Draft_Shape2DView'
00:09:32  Unknown command 'Draft_LayerManager'
00:09:32  Unknown command 'Draft_AddNamedGroup'
00:09:32  Unknown command 'Draft_AddToGroup'
00:09:32  Unknown command 'Draft_SelectGroup'
00:09:32  Unknown command 'Draft_AddConstruction'
00:09:32  Unknown command 'Draft_ToggleDisplayMode'
00:09:32  Unknown command 'Draft_WorkingPlaneProxy'
00:09:32  Unknown command 'Draft_Snap_Lock'
00:09:32  Unknown command 'Draft_Snap_Endpoint'
00:09:32  Unknown command 'Draft_Snap_Midpoint'
00:09:32  Unknown command 'Draft_Snap_Center'
00:09:32  Unknown command 'Draft_Snap_Angle'
00:09:32  Unknown command 'Draft_Snap_Intersection'
00:09:32  Unknown command 'Draft_Snap_Perpendicular'
00:09:32  Unknown command 'Draft_Snap_Extension'
00:09:32  Unknown command 'Draft_Snap_Parallel'
00:09:32  Unknown command 'Draft_Snap_Special'
00:09:32  Unknown command 'Draft_Snap_Near'
00:09:32  Unknown command 'Draft_Snap_Ortho'
00:09:32  Unknown command 'Draft_Snap_Grid'
00:09:32  Unknown command 'Draft_Snap_WorkingPlane'
00:09:32  Unknown command 'Draft_Snap_Dimensions'
00:09:32  Unknown command 'Draft_ToggleGrid'
00:09:32  Unknown command 'Draft_Fillet'
00:09:32  Unknown command 'Draft_Circle'
00:09:32  Unknown command 'Draft_Ellipse'
00:09:32  Unknown command 'Draft_Rectangle'
00:09:32  Unknown command 'Draft_Polygon'
00:09:32  Unknown command 'Draft_BSpline'
00:09:32  Unknown command 'Draft_Point'
00:09:32  Unknown command 'Draft_Facebinder'
00:09:32  Unknown command 'Draft_ShapeString'
00:09:32  Unknown command 'Draft_Text'
00:09:32  Unknown command 'Draft_Dimension'
00:09:32  Unknown command 'Draft_Label'
00:09:32  Unknown command 'Draft_AnnotationStyleEditor'
00:09:32  Unknown command 'Draft_Move'
00:09:32  Unknown command 'Draft_Rotate'
00:09:32  Unknown command 'Draft_Scale'
00:09:32  Unknown command 'Draft_Mirror'
00:09:32  Unknown command 'Draft_Offset'
00:09:32  Unknown command 'Draft_Trimex'
00:09:32  Unknown command 'Draft_Stretch'
00:09:32  Unknown command 'Draft_Clone'
00:09:32  Unknown command 'Draft_Edit'
00:09:32  Unknown command 'Draft_SubelementHighlight'
00:09:32  Unknown command 'Draft_Join'
00:09:32  Unknown command 'Draft_Split'
00:09:32  Unknown command 'Draft_Upgrade'
00:09:32  Unknown command 'Draft_Downgrade'
00:09:32  Unknown command 'Draft_WireToBSpline'
00:09:32  Unknown command 'Draft_Draft2Sketch'
00:09:32  Unknown command 'Draft_Slope'
00:09:32  Unknown command 'Draft_FlipDimension'
00:09:32  Unknown command 'Draft_Shape2DView'
00:09:32  Unknown command 'Draft_SetStyle'
00:09:32  Unknown command 'Draft_ApplyStyle'
00:09:32  Unknown command 'Draft_Layer'
00:09:32  Unknown command 'Draft_LayerManager'
00:09:32  Unknown command 'Draft_AddNamedGroup'
00:09:32  Unknown command 'Draft_AddToGroup'
00:09:32  Unknown command 'Draft_SelectGroup'
00:09:32  Unknown command 'Draft_ToggleConstructionMode'
00:09:32  Unknown command 'Draft_AddConstruction'
00:09:32  Unknown command 'Draft_ToggleDisplayMode'
00:09:32  Unknown command 'Draft_ToggleGrid'
00:09:32  Unknown command 'Draft_SelectPlane'
00:09:32  Unknown command 'Draft_WorkingPlaneProxy'
00:09:32  Unknown command 'Draft_Heal'
00:09:32  Unknown command 'Draft_ShowSnapBar'
00:09:33  Traceback (most recent call last):
00:09:33    File "/usr/share/freecad-daily/Mod/Draft/draftutils/init_draft_statusbar.py", line 247, in init_draft_statusbar_snap
00:09:33      snap_widget.addAction(Gui.Command.get("Draft_ToggleGrid").getAction()[0])
00:09:33  AttributeError: 'NoneType' object has no attribute 'getAction'
```

Most of the buttons have disappeared and it basically doesn't do anything now.

### Full version info

```shell
OS: Ubuntu 22.04 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36097 (Git)
Build type: Release
Branch: main
Hash: 14d2915e6c5828a697921cc9144ef368f8dd0e2f
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Installed mods: 
  * Curves 0.6.23
  * Help 1.0.3
  * lattice2 1.0.0
  * A2plus 0.4.63
  * CurvedShapes 1.0.5
```
```

### Subproject(s) affected?

Draft

### Anything else?

This happens with 0.22dev and 0.21, both installed with my package manager. I thought installing the dev version would fix it, but it did not.

I am using these PPAs.

```
ppa:freecad-maintainers/freecad-stable
ppa:freecad-maintainers/freecad-daily
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-02-24:
@adrianinsaval AFAIK you handle the PPAs. Can you have a look at this issue? Thanks.

Roy-043, 2024-02-24:
@FalcoGer Can you try the AppImage?
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

FalcoGer, 2024-02-24:
I would prefer to keep using the package manager. But the appimage works as expected.

adrianinsaval, 2024-03-06:
I can't reproduce this so it's not an inherent problem of the ppa distribution. I would expect some sort of conflict in your system. Are you still experiencing this if you use the ppa @FalcoGer ?

FalcoGer, 2024-03-08:
Yes, I still get the same errors, updated just now. Probably missing or wrong version dependencies.

```text
19:34:50  <built-in function SoFieldContainer_getField> returned a result with an exception setError: Initializing one or more of the Draft modules failed, Draft will not work as expected.
19:34:50  Unknown command 'Draft_Fillet'
19:34:50  Unknown command 'Draft_Circle'
19:34:50  Unknown command 'Draft_Ellipse'
19:34:50  Unknown command 'Draft_Rectangle'
19:34:50  Unknown command 'Draft_Polygon'
19:34:50  Unknown command 'Draft_BSpline'
19:34:50  Unknown command 'Draft_Point'
19:34:50  Unknown command 'Draft_Facebinder'
19:34:50  Unknown command 'Draft_ShapeString'
19:34:50  Unknown command 'Draft_Text'
19:34:50  Unknown command 'Draft_Dimension'
19:34:50  Unknown command 'Draft_Label'
19:34:50  Unknown command 'Draft_AnnotationStyleEditor'
19:34:50  Unknown command 'Draft_Move'
19:34:50  Unknown command 'Draft_Rotate'
19:34:50  Unknown command 'Draft_Scale'
19:34:50  Unknown command 'Draft_Mirror'
19:34:50  Unknown command 'Draft_Offset'
19:34:50  Unknown command 'Draft_Trimex'
19:34:50  Unknown command 'Draft_Stretch'
19:34:50  Unknown command 'Draft_Clone'
19:34:50  Unknown command 'Draft_Edit'
19:34:50  Unknown command 'Draft_SubelementHighlight'
19:34:50  Unknown command 'Draft_Join'
19:34:50  Unknown command 'Draft_Split'
19:34:50  Unknown command 'Draft_Upgrade'
19:34:50  Unknown command 'Draft_Downgrade'
19:34:50  Unknown command 'Draft_WireToBSpline'
19:34:50  Unknown command 'Draft_Draft2Sketch'
19:34:50  Unknown command 'Draft_Slope'
19:34:50  Unknown command 'Draft_FlipDimension'
19:34:50  Unknown command 'Draft_Shape2DView'
19:34:50  Unknown command 'Draft_LayerManager'
19:34:50  Unknown command 'Draft_AddNamedGroup'
19:34:50  Unknown command 'Draft_AddToGroup'
19:34:50  Unknown command 'Draft_SelectGroup'
19:34:50  Unknown command 'Draft_AddConstruction'
19:34:50  Unknown command 'Draft_ToggleDisplayMode'
19:34:50  Unknown command 'Draft_WorkingPlaneProxy'
19:34:50  Unknown command 'Draft_Snap_Lock'
19:34:50  Unknown command 'Draft_Snap_Endpoint'
19:34:50  Unknown command 'Draft_Snap_Midpoint'
19:34:50  Unknown command 'Draft_Snap_Center'
19:34:50  Unknown command 'Draft_Snap_Angle'
19:34:50  Unknown command 'Draft_Snap_Intersection'
19:34:50  Unknown command 'Draft_Snap_Perpendicular'
19:34:50  Unknown command 'Draft_Snap_Extension'
19:34:50  Unknown command 'Draft_Snap_Parallel'
19:34:50  Unknown command 'Draft_Snap_Special'
19:34:50  Unknown command 'Draft_Snap_Near'
19:34:50  Unknown command 'Draft_Snap_Ortho'
19:34:50  Unknown command 'Draft_Snap_Grid'
19:34:50  Unknown command 'Draft_Snap_WorkingPlane'
19:34:50  Unknown command 'Draft_Snap_Dimensions'
19:34:50  Unknown command 'Draft_ToggleGrid'
19:34:50  Unknown command 'Draft_Fillet'
19:34:50  Unknown command 'Draft_Circle'
19:34:50  Unknown command 'Draft_Ellipse'
19:34:50  Unknown command 'Draft_Rectangle'
19:34:50  Unknown command 'Draft_Polygon'
19:34:50  Unknown command 'Draft_BSpline'
19:34:50  Unknown command 'Draft_Point'
19:34:50  Unknown command 'Draft_Facebinder'
19:34:50  Unknown command 'Draft_ShapeString'
19:34:50  Unknown command 'Draft_Text'
19:34:50  Unknown command 'Draft_Dimension'
19:34:50  Unknown command 'Draft_Label'
19:34:50  Unknown command 'Draft_AnnotationStyleEditor'
19:34:50  Unknown command 'Draft_Move'
19:34:50  Unknown command 'Draft_Rotate'
19:34:50  Unknown command 'Draft_Scale'
19:34:50  Unknown command 'Draft_Mirror'
19:34:50  Unknown command 'Draft_Offset'
19:34:50  Unknown command 'Draft_Trimex'
19:34:50  Unknown command 'Draft_Stretch'
19:34:50  Unknown command 'Draft_Clone'
19:34:50  Unknown command 'Draft_Edit'
19:34:50  Unknown command 'Draft_SubelementHighlight'
19:34:50  Unknown command 'Draft_Join'
19:34:50  Unknown command 'Draft_Split'
19:34:50  Unknown command 'Draft_Upgrade'
19:34:50  Unknown command 'Draft_Downgrade'
19:34:50  Unknown command 'Draft_WireToBSpline'
19:34:50  Unknown command 'Draft_Draft2Sketch'
19:34:50  Unknown command 'Draft_Slope'
19:34:50  Unknown command 'Draft_FlipDimension'
19:34:50  Unknown command 'Draft_Shape2DView'
19:34:50  Unknown command 'Draft_SetStyle'
19:34:50  Unknown command 'Draft_ApplyStyle'
19:34:50  Unknown command 'Draft_Layer'
19:34:50  Unknown command 'Draft_LayerManager'
19:34:50  Unknown command 'Draft_AddNamedGroup'
19:34:50  Unknown command 'Draft_AddToGroup'
19:34:50  Unknown command 'Draft_SelectGroup'
19:34:50  Unknown command 'Draft_ToggleConstructionMode'
19:34:50  Unknown command 'Draft_AddConstruction'
19:34:50  Unknown command 'Draft_ToggleDisplayMode'
19:34:50  Unknown command 'Draft_ToggleGrid'
19:34:50  Unknown command 'Draft_SelectPlane'
19:34:50  Unknown command 'Draft_WorkingPlaneProxy'
19:34:50  Unknown command 'Draft_Heal'
19:34:50  Unknown command 'Draft_ShowSnapBar'
19:34:51  Traceback (most recent call last):
19:34:51    File "/usr/share/freecad-daily/Mod/Draft/draftutils/init_draft_statusbar.py", line 247, in init_draft_statusbar_snap
19:34:51      snap_widget.addAction(Gui.Command.get("Draft_ToggleGrid").getAction()[0])
19:34:51  AttributeError: 'NoneType' object has no attribute 'getAction'
```

Which python/pip packages do I need?

adrianinsaval, 2024-03-08:
can you uninstall and purge freecad-daily, freecad, opencascade and pivy and coin3d and the reinstall? THe package should install everything you need from the repo. If possible clear everything you had installed with pip as that often creates conflicts.

FalcoGer, 2024-03-08:
I can't find opencascade and coin3d. and i can only find python3-pivy.

FalcoGer, 2024-04-27:
switched to flatpak after dist upgrade to 24.04. that works.

adrianinsaval, 2024-04-28:
I was never able to reproduce this problem, I'll assume it was some weird problem in your system since you don't have it anymore to keep investigating. I will update the daily ppa to have a package for 24.04 soon, you may test to see if the issue persists then if you're interested.

closed 2024-04-28 by adrianinsaval