FreeCAD/FreeCAD #12578: FreeCAD ARM 36117 hard crash when deleting Sketcher line segment

Bindelreit, 2024-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When deleting a particular line segment in a sketch, FreeCAD crashes. Running on Apple ARM processor.

Happens with the developer ARM 22.0.36117 and the stable 21.2 version. 

Screen recording attached as well as the bug report and the FCStd file.

### Full version info

```shell
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36117 (Git)
Build type: Release
Branch: main
Hash: 8a3407aee6dff8218282d9cdc3f4e32e7ad958d4
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Curves 0.6.12
```
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/6846928/ed60d3b8-2c62-4972-a949-ad238b7c069b

[FreeCAD crash report.txt](https://github.com/FreeCAD/FreeCAD/files/14391759/FreeCAD.crash.report.txt)

[multi-material tests2.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14391762/multi-material.tests2.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Bindelreit, 2024-02-24:
UPDATE: Crash occurs when the ellipse is present. Deleting the ellipse first or at the same time as the horizontal line, allows deletion of the horizontal line and prevents the crash. See video and crash log. Same FCStd file as previously uploaded.

[FreeCAD Crash Report Issue 12578.txt](https://github.com/FreeCAD/FreeCAD/files/14394138/FreeCAD.Crash.Report.Issue.12578.txt)

https://github.com/FreeCAD/FreeCAD/assets/6846928/0175e55e-afaa-489d-98be-218e61fc04c3

luzpaz, 2024-03-13:
@JohnOCFII do you have a moment to confirm ?

JohnOCFII, 2024-03-13:
> @JohnOCFII do you have a moment to confirm ?

I'm not able to duplicate the crash.  I've tried deleting those three elements (line, arc, line)  in different order, or two of the three at the same time, and never experienced a crash.  I **did** receive the following messages in the `Report View` window, which might lend a clue.

```
10:44:18  ViewProviderSketch: Delete: Selection not restricted to one sketch and its subelements
10:44:21  ViewProviderSketch: Delete: Selection not restricted to one sketch and its subelements
10:44:34  AdditivePipe009: A fatal error occurred when making the pipe
10:44:34  AdditivePipe009: A fatal error occurred when making the pipe
```
```
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * OpenLight_Beta 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
```

maxwxyz, 2024-10-07:
@Bindelreit can you test on the latest RC / dev version please.
I am also unable to reproduce on Windows
```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```

luzpaz, 2024-10-16:
@Bindelreit :point_up_2:

luzpaz, 2024-10-24:
@Bindelreit ping

maxwxyz, 2024-11-01:
I'll close this as it could not be reproduced and there was no further feedback. Feel free to add a comment later and we can reopen.

closed 2024-11-01 by maxwxyz