FreeCAD/FreeCAD #12584: [Problem] UI/UX: Origin planes are too intrusive and only selectable at the edges

maxwxyz, 2024-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a double sided sword. The current default origin / axis system datum planes (XY-plane, XZ-plane, YZ-plane) of each body or part are displayed as huge planes but one can only select them at the outer edge. In assemblies or when dealing with multiple bodies, the selection of the edge is the only way to work with them, as they are all overlapped on top of each other.
In this demo I have an assembly with 3 simple parts and it is already not usable or distinguishable.

https://github.com/FreeCAD/FreeCAD/assets/6246609/441d6be2-9a71-4cee-a1b7-301ec5ed585b

Ideally the origin planes could be displayed less intrusive and also keep all relevant information including the positive axis indication (e.g. which way is +x). A smaller origin UI works also with distinguishing different part origins and keeps the same select-ability as today.

Examples from other CAD software.
CATIA:
[image]
In assemblies an easier overview for the user: 
[image]
Fusion360:
[image]
Solidworks:
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36082 (Git)
Build type: Release
Branch: main
Hash: 7a025e4e60a607bdbdd1599627ce13bb8eeda071
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.11
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.1
```

### Subproject(s) affected?

Core

### Anything else?

Maybe a new origin UI could be unified with the local coordinate system of joints in the Assembly WB (JCS / LCS) as well, to be visually in line throughout the application.

Even the indication of user-created datum planes (https://github.com/FreeCAD/FreeCAD/pull/12296) and sketches (https://github.com/FreeCAD/FreeCAD/issues/12458) could be unified with a new origin appearance as well.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-24:
@obelisk79 maybe also worth discussing in the DWG: 
Changing the default UI/UX for the origin and using a unified solution for planes, sketches, local coordinate systems (JCS) in assemblies (@PaddleStroke ).
As mentiones above, there are already discussions and ideas in https://github.com/FreeCAD/FreeCAD/pull/12296  or https://github.com/FreeCAD/FreeCAD/issues/12458

obelisk79, 2024-02-24:
I'll have to start this conversation later with the group later. Some time ago on the forum I believe there was what I thought a great proposal. To use planes with a center cutout, but the entire plane area was selectable. They would look similar to this rudimentary drawing I made:
[image]

PaddleStroke, 2024-02-26:
Imo the best solution is : 
- Smaller planes, so that they do not get in the way of other faces and so on. Actually the planes do not scale with zoom level. So with default zoom level they are too big, but if you zoom out they become too small. They should scale with zoom level (the same way JCS in assembly scale with zoom).
- All the face is selectable an not only the edge and label.
- Draw planes only in their positive areas. Basically like fusion/catia screenshots above. So they do not overlap and are simpler to understand.

maxwxyz, 2024-02-26:
I prefer the SolidWorks or CATIA solution: the axis names are displayed with a letter and it is easy to click on each one, also for the planes. The smaller ones with a fully selectable face could get messy when multiple parts are near each other, as there is no way of clicking through one LCS if you need the other.

pierreporte, 2024-02-26:
@maxwxyz Your screenshots may be misleading for those who don’t use Catia. You show images of the coordinate systems. The actual planes are little squares (the size is a program setting) without a label and their size remains the same regardless of the zoom level (the size is always a fixed pixel size). While I would personally be in favor of this as I find that large planes get in the way (even as simple unfilled rectangles), it may not be the case for everyone.

This is a suggestion based on experience:

* Planes are drawn with an thin opaque border and a transparent background.
* Planes have a constant pixel size by default.
* There is a parameter for each one to extend them to the container bounding box (sometimes it’s a pain to pick a plane when you’re working at the other size of a large part).
* There is a program settings to make the plane extended by default.
* Planes have a label attached, but not as big as today.
* The coordinate system is displayed on hover or selection by default.
* There is a program setting to always display the coordinate system.
* The coordinate system axis have labels.

------------------

*Default size:*

[image]

*Custom size:*

[image]

obelisk79, 2024-02-27:
If we are going to pursue one, I think the most visually clear approach is not what is shown in the catia screenshots. Rather a hybrid of what is seen in the solidworks and fusion images. Ideally, I think taking the same approach as Fusion and adding the arrowpoints and letters from solidworks would make a very effective visual representation.

maxwxyz, 2024-02-28:
I agree, plus also what @PaddleStroke and @pierreporte mentioned: in those applications the axis cross stays the same pixel size and does not scale / is not displayed larger when you zoom in. That would also help for a better overview and less clutter in the 3D view. 
So maybe the solution is to define a preference setting for the size of the origin / planes.

And the new axis cross should ideally then be used to be displayed not only for the origin of parts but also for sketches and datums (visible by default) and the local coordinate systems for joints in assemblies (JCS).

maxwxyz, 2024-02-28:
@qewer33 / @kadet1090 maybe you can make weigh in, especially with the transform manipulator changes.

maxwxyz, 2024-03-10:
I'll add the remarks from the FreeCAD Day 2024 Complaint Session:
**Reference objects such as datum lines and planes should be available globally to all WB.**
Reference objects such as datum planes, axes lines and axis systems should be available globally to all workbenches.
Consistent behavior and display of them is key.

Pendelf, 2024-09-14:
When you select a plane to create a sketch in PartDesign when you zoom in, the planes become very small

https://github.com/user-attachments/assets/5e98cc81-7ef1-4325-a1c4-ff3d50a35568

It may be worth making the planes scale depending on the zoom

https://github.com/user-attachments/assets/32a2032a-01f6-42ec-915e-76967fe287f5

PaddleStroke, 2024-09-18:
I am planning to start working on this issue since I was in the process of making a 'Marker tool' in assembly, which would only further this issue by adding yet another LCS tool/object.

So my current plan now is to modify the current App::Origin object such that it can do the job of PartDesign datums and assembly markers.

The plan : 
- Add origin point to the origin object since it's lacking.
- Rename App::Origin object to 'App::LocalCoordinateSystem'
- Change the graphics : 
- - Smaller size + resize with zoom
- - Like fusion but with arrows and letters X Y Z
- - Draw on top so that it should not be hidden by solids
- Add a 'Reference' property such that it can be attached.
- Add a 'findPlacement(subNames)' function to App::DocumentObject that returns identity plc by default. This function can be reimplemented, in particular in Part so that attachement can work correctly.
- Make it so that the LCS becomes a tree-child of the referenced object.
- Add a task dialog to the LCS edition, that let you modify its attachement. The task should provide full attachement capability while being simple enough.
- Add a AttachementOffset property.
- Add datum objects tools in the 'structure' toolbar
- Make sure PartDesign can use the core datums.
- Deprecate PartDesign datums tools.

PaddleStroke, 2024-09-18:
@wwmayer I'd like to know what you think about this please.

adrianinsaval, 2024-09-18:
what happens with the existing plane and line objects? maybe you should keep those names as they are and name your new objects something like `App::Datum[Line,Plane,Point]` to reduce file porting complexity

PaddleStroke, 2024-09-18:
That would certainly be simpler. However why keeping these current App::Line/Plane? Searching through the code I see few references to App::Line, but nothing we can't handled.
I feel the best solution is to overwrite them. It will leave the code cleaner.

adrianinsaval, 2024-09-18:
if you think migrating documents will be trivial I don't see a problem, just something to keep in mind

maxwxyz, 2024-09-18:
@FreeCAD/design-working-group @FreeCAD/cad-advisory-group please weigh in:

> I am planning to start working on this issue since I was in the process of making a 'Marker tool' in assembly, which would only further this issue by adding yet another LCS tool/object.
>
> So my [...]

pierreporte, 2024-09-18:
I’m basing my opinion on the current origin object:

[image]

> * Add origin point to the origin object since it's lacking.

Yes please.

> * Rename `App::Origin` object to `App::LocalCoordinateSystem`

It would make sense with the actual origin becoming a point.

> * Change the graphics :
>
>   * Smaller size + resize with zoom
>   * Like fusion but with arrows and letters X Y Z
>   * Draw on top so that it should not be hidden by solids

Yes for small size. The screen size of the labels definitely need to stay the same regardless of zoom level. If axis and planes have a constant screen size, it shouldn’t be limited to the origin object. It could even be an attribute (keep screen size constant or fit to object).

Yes for arrows and letters, but I think base planes should be displayed as in Catia or Solidworks, with some sort of corner, to avoid the confusion with actual XY, YZ and ZX planes. Also the letters would allow a monochrome style, to reduce amount of colors (it would fit my new sketcher style proposal that doesn’t have colors for axis).

We already have dashed lines when they are hidden but drawing on top of everything is perfectly fine.

> * Add a 'Reference' property such that it can be attached.
> * Add a 'findPlacement(subNames)' function to App::DocumentObject that returns identity plc by default. This function can be reimplemented, [...]

I don’t understand this. Is it about internals?

> * Add a task dialog to the LCS edition, that let you modify its attachement. The task should provide full attachement capability while being simple enough.

Yes.

> * Add a AttachementOffset property.
> * Add datum objects tools in the 'structure' toolbar
> * Make sure PartDesign can use the core datums.
> * Deprecate PartDesign datums tools.

Yes please. It should have been the case from the beginning. It will allow to easily add datum in assemblies as well. They will need to be in the toolbar.

PaddleStroke, 2024-09-20:
Making good progress
[image]

pierreporte, 2024-09-20:
@PaddleStroke It looks great! Can you make the label in the screen plane?

kadet1090, 2024-09-20:
Nice work!

Few notes:
1. You should use So3DAnnotation so it takes care of depth buffer and proper ordering of planes when rendered.
2. Axes should be a bit longer than planes so they would be easier to select (see the Y axis on your screen)
3. You should use theme colors from user preferences
4. I'd center axis labels at the end of line
5. (this is totally personal opinion) planes do not need such visible border - I'd make them semi-transparent (but not as much as the plane itself)

PaddleStroke, 2024-09-20:
This PR is already getting out of hand and is modifying a lot of files. So for some rendering improvements they can be added after.
- labels parallel to screen and at the axis end : +1 but I don't know how to do this. Later PR
- So3DAnnotation : don't know how to do this. Later PR
- Longer axis : They are already longer than planes. But I can make it even more.
- Themes colors: can you tell me the pref path?
- plane rendering : +1 I'll check if it's easy to do.

kadet1090, 2024-09-20:
If you are putting SoAnnotation in in your code just use So3DAnnotation, nothing else is needed - if you can push code I can submit a patch for you

PaddleStroke, 2024-09-20:
There's one SoAnnotation for the hidden features. What include to use So3DAnnotation? Just changing to :
`    auto hidden = new So3DAnnotation();`
is unknown identifier.

Here's how it looks with smaller planes with transparent borders
[image]

kadet1090, 2024-09-20:
https://github.com/FreeCAD/FreeCAD/blob/8b9f5bdc4f4bd3cf4a440365fcc7cbcd2afe8580/src/Gui/So3DAnnotation.h#L22 this is the file you need to include. But let's waif for the PR - I'll check if this is really an issue and submit you diff if needed in comment.

pierreporte, 2024-09-20:
I will create an issue for labels.

Does your PR makes them the same size regardless of zoom level? If not, do you plan to do is it also for later?

kadet1090, 2024-09-20:
As for the axis colors you can obtain them from ViewParams: https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/ViewParams.h#L66-L68

Like so: https://github.com/FreeCAD/FreeCAD/blob/8b9f5bdc4f4bd3cf4a440365fcc7cbcd2afe8580/src/Gui/ViewProviderDragger.cpp#L142-L146

PaddleStroke, 2024-09-20:
@kadet1090 PR is up already if you want to look : https://github.com/FreeCAD/FreeCAD/pull/16675
It's not done.

PaddleStroke, 2024-09-20:
@pierreporte yes it stays the same size.

pierreporte, 2024-09-20:
@PaddleStroke BTW what is this assembly marker tool?

PaddleStroke, 2024-09-26:
A WIP: https://github.com/Ondsel-Development/FreeCAD/tree/asm_markers
The idea was to be able to create markers objects. So the same as when you create a joint, but it would just create a LCS object instead of a joint. With the same UI as the joint creation tool: you hover the solid and get a preview.

The idea was to support the workflow LCS -> then joint creation. Also for MbD it'll be usefull.

But while I was working on it, I realised that I was reimplementing yet another LCS tool. Making this issue even worse. So I decided to take a step back and fix this so that assembly can use core datums instead of a new object type.

closed 2024-12-13 by chennes