FreeCAD/FreeCAD #12586: [Problem] FEM requires OpenMP 5.0 (e.g. GCC >= 9)

StefanBruens, 2024-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

GCC 7.5 should be sufficient, according to https://wiki.freecad.org/FreeCAD_1.0_Development_Cycle

FEM now uses range-base for loops in combination with `#pragma parallel for`, which is allowed only with OpenMP 5.0 or later (see e.g. https://stackoverflow.com/questions/17848521/using-openmp-with-c11-range-based-for-loops ).

The relevant change is https://github.com/FreeCAD/FreeCAD/commit/009f9f2222e7e590f8391b5a62f782893c3928dc

@wwmayer 

### Full version info

```shell
Current git (2024-02-24)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-02-24:
Are there only these four cases where omp is used together with range-based `for` loops? I think it shouldn't be a problem to switch back to the old-style `for` loops for them.

StefanBruens, 2024-02-24:
> Are there only these four cases where omp is used together with range-based `for` loops? I think it shouldn't be a problem to switch back to the old-style `for` loops for them.

AFAICS yes.

closed 2024-02-24 by wwmayer

dzid26, 2024-04-06:
Can we set OpenMP version `find_package(OpenMP 5.0)` ? in https://github.com/FreeCAD/FreeCAD/blob/d1930fc7e6fc35c413f9f39d22f2ea28fca827c7/src/Mod/Fem/App/CMakeLists.txt#L199-L202

For me OpenMP 2.0 was picked up and caused bunch of errors:
> C:\Users\dzidm\Downloads\FreeCAD\src\Mod\Fem\App\FemMesh.cpp
[build] C:\Users\dzidm\Downloads\FreeCAD\src\Mod\Fem\App\FemMesh.cpp(867): error C3016: 'i': index variable in OpenMP 'for' statement must have signed integral type
[build] C:\Users\dzidm\Downloads\FreeCAD\src\Mod\Fem\App\FemMesh.cpp(888): fatal error C1903: unable to recover from previous error(s); stopping compilation 

I don't know why can't it find version 5.0 though, since it is present in conda, but I guess that's a separate issue.

StefanBruens, 2024-04-12:
> Can we specify OpenMP version `find_package(OpenMP 5.0)` ? 

As was already mentioned here, OpenMP 5.0 is only available with e.g. GCC 9, but the current baseline is GCC 7.5. OpenMP 4 is quite sufficient for FreeCADs current needs.

The issue you face is apparently a local build system setup error, if OpenMP 5 can not be found even if it is there, then raising the minimum version won't help.

dzid26, 2024-04-12:
Undestood. So can't we require version >=4.0? 

My issue was that the build was failing beacuse it was finding version 2.0 (visial studio 2019). Wrong version should not be accepted.

commit e57a1fe0 (2024-02-24): Fixes #12586: FEM requires OpenMP 5.0 (e.g. GCC >= 9)