FreeCAD/FreeCAD #12587: [Problem] Sketcher build fails with "-Werror=return-type"

StefanBruens, 2024-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

```
[ 2642s] /home/abuild/rpmbuild/BUILD/FreeCAD-0.22.0~g20240223.628bad452/src/Mod/Sketcher/Gui/TaskSketcherElements.cpp: In member function â€˜bool SketcherGui::ElementItem::isGeometrySelected(Sketcher::PointPos) constâ€™:
[ 2642s] /home/abuild/rpmbuild/BUILD/FreeCAD-0.22.0~g20240223.628bad452/src/Mod/Sketcher/Gui/TaskSketcherElements.cpp:245:5: error: control reaches end of non-void function [-Werror=return-type]
[ 2642s]   245 |     }
[ 2642s]       |     ^
```

The problematic change was introduced with
https://github.com/FreeCAD/FreeCAD/commit/da6a4fe57b8ba60fa8e3c80b3f0d47132ee8c087

The compiler does not know, without being told explicitly, that the enum value is restricted to 4 different value, it has to assume all values from the underlying type are possible.

This is a common problem; unfortunately, a portable solution only exists with C++23:  
https://en.cppreference.com/w/cpp/utility/unreachable

Until FreeCAD uses C++23 it may provide a macro (e.g `freecad_unreachable`) which uses the portable "Possible implementaton" from cppreference.

### Full version info

```shell
Current git (2024-02-24)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

StefanBruens, 2024-02-24:
https://github.com/FreeCAD/FreeCAD/commit/da6a4fe57b8ba60fa8e3c80b3f0d47132ee8c087#diff-580e506086306907cd5cd0319adafdb61145a926b37a70f097b8a1b915f63db1R228-R241

maxwxyz, 2024-02-24:
@kadet1090

kadet1090, 2024-02-24:
Do we actually compile with this flag? If yes, in what scenario? This commit builds fine in the CI, was introduced some time ago without anyone noticing it and neither my IDE nor linter in CI did not complain about the issue. If we care about compilation with this flag and treat it as a bug we should adjust CI to properly find such issues

@chennes you may have some valuable input on that matter?

FlachyJoe, 2024-02-24:
A crash can occurs if someone cast a invalid value to `Sketcher::PointPos`. The case is improbable but we can manage it.

We can add a default case:
```cpp
    bool isGeometrySelected(Sketcher::PointPos pos) const
    {
        switch (pos) {
            case Sketcher::PointPos::none:
                return isLineSelected;
            case Sketcher::PointPos::start:
                return isStartingPointSelected;
            case Sketcher::PointPos::end:
                return isEndPointSelected;
            case Sketcher::PointPos::mid:
            default:
                return isMidPointSelected;
        }
    }
```

or return false for non handled:
```cpp
    bool isGeometrySelected(Sketcher::PointPos pos) const
    {
        switch (pos) {
            case Sketcher::PointPos::none:
                return isLineSelected;
            case Sketcher::PointPos::start:
                return isStartingPointSelected;
            case Sketcher::PointPos::end:
                return isEndPointSelected;
            case Sketcher::PointPos::mid:
                return isMidPointSelected;
        }
        //invalid pos
        return false;
    }
```

kadet1090, 2024-02-24:
Personally I think that we should go with something like "Assert Unreached" (which should not be checked in release builds) and the FreeCAD should crash in that case. Failing silently seems much worse in that case for me as it would be much harder to find on development stage.

wwmayer, 2024-02-27:
> Do we actually compile with this flag?

With MSVC I got a warning that was fixed with #12567.

With clang (using -Wall -Wextra -Wpedantic) I don't get a warning because the switch handled all four cases defined in PointPos. But one could cast an arbitrary int value to PointPos and pass it to isGeometrySelected() then the case won't be handled and causes UB. gcc complains about this without having to set any explicit warn level.

Here is a little test application
```
#include <iostream>

enum class PointPos : int
{
    none = 0,
    start = 1,
    end = 2,
    mid = 3
};

int test(PointPos pos)
{
    switch (pos) {
        case PointPos::none:
            return 0;
        case PointPos::start:
            return 1;
        case PointPos::end:
            return 2;
        case PointPos::mid:
            return 3;
    }
}

int main()
{
    std::cout << test(PointPos::none) << '\n'
              << test(PointPos::start) << '\n'
              << test(PointPos::end) << '\n'
              << test(PointPos::mid) << '\n';
    std::cout << test(PointPos(7)) << '\n';
}
```

When building it with gcc: `g++ main.cpp` raises the warning `main.cpp:23:1: warning: control reaches end of non-void function [-Wreturn-type]`. Executing the binary shows this output where the last value is a random number:
```
0
1
2
3
-1364156352 
```

When building it with clang: `clang++ -Wall -Wextra -Wpedantic main.cpp` no warning is raised. When running the application then the output is:
```
0
1
2
3
Illegal instruction (core dumped)
```

Changing the test function fixes the UB:
```
int test(PointPos pos)
{
    switch (pos) {
        case PointPos::none:
            return 0;
        case PointPos::start:
            return 1;
        case PointPos::end:
            return 2;
        case PointPos::mid:
            return 3;
        default:
            return -1;
    }
}
```

wwmayer, 2024-03-15:
This can be closed as it's already fixed.

closed 2024-03-15 by wwmayer