FreeCAD/FreeCAD #12591: [Problem] Expression Editor dialog box partially offscreen

iAmVaeVictus, 2024-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When editing formulas for a Length constraint, when the formula is somewhat long, and the constraint item in the drawing is near the right edge of the screen, the Expression Editor dialog ends up halfway off the right edge of the screen, requiring me to grab and drag the box over to see the end of the expression. (Screen shot provided). 

It would be nice to have for this to adjust to the edge of the screen instead of going off the screen. Not a bug, but an enhancement obviously! 

Great work folks, keep it up! 
[image]

### Full version info

```shell
OS: macOS 13.6.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-02-24:
Related: #12518.

iAmVaeVictus, 2024-02-24:
> Related: #12518.

Yes. But not a duplicate.