FreeCAD/FreeCAD #12592: [Problem] Sketcher trimming periodic B-spline by knots results in deformed geometry

Roy-043, 2024-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trimming a periodic B-spline by knots results in deformed geometry:

Open [spline-trim.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14393737/spline-trim.FCStd.zip) (remove the zip extension).
Open the sketch.
Trim the right side of the periodic B-spline between the 2 lines.
Result: O-shape becomes a self-crossing Y-shape.

V0.21 does not have this issue.

[image]

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36117 (Git)
Build type: Release
Branch: main
Hash: 8a3407aee6dff8218282d9cdc3f4e32e7ad958d4
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-02:
@AjinkyaDahale FYI

AjinkyaDahale, 2024-05-31:
The only changes in `SketchObject::trim()` in 2023 or later are 987b4bda2ac8570969dc07fa887bda420f77e441 and a32851073d986f5211ab6bef82d26af9a8e185c5. If their reversal doesn't fix the issue the problem is likely deeper.

AjinkyaDahale, 2024-05-31:
@PaddleStroke could you confirm if this problem still exists after #14079? I have overwritten that build and getting it back will take a long time.

AjinkyaDahale, 2024-06-02:
#14461 resolves the issue with periodic B-spline, but turns out the very adjacent non-periodic B-spline is not perfect either. If you try to trim the portion between the same lines, one of the remaining pieces changes shape.

maxwxyz, 2024-06-24:
@Roy-043 could you confirm that this issue is fixed now?

Roy-043, 2024-06-25:
I don't compile myself. The PR has to be merged first before I can test it.

PR #14461 (2024-07-01): [Part][Sketcher] Fix some B-spline trimming issues
Fixes #12592.

~There is still a problem with the other (non-periodic) B-spline shown in that issue's example.~

If anyone wants to check it out, this issue was introduced in 728a880078f04e24bfe5ed215c95c4032f073683.
closed 2024-07-01 by chennes