FreeCAD/FreeCAD #12595: In Workbench assembly, forcing the user to create a Ground on an object is unnecessary and only limits the software

luisDias89, 2024-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The issue is that the assembler is forcing the user to ground a body before starting to create joins/mates, which is incorrect. I'll first demonstrate how it's done in a professional software, and then I'll explain what's happening in FreeCAD that makes the entire assembly buggy:

Freecad BUG: It shouldn't be necessary to have a grounded object to create a (mate/join).
https://github.com/FreeCAD/FreeCAD/assets/48370099/1720d72f-db5b-4214-ba6a-5b70729674f3

I'll provide a video example of another software where it's not required, it simple and so easy:
https://github.com/FreeCAD/FreeCAD/assets/48370099/530f6ebf-5604-45cd-963a-077107dc9ec3

Please don't take my criticisms the wrong way. I've attempted to migrate to FreeCAD several times, but I work with large assemblies, and no Workbench is mature enough for that. I'll try to help with my suggestions in the coming months, as I have over 10 years of experience in CAD design and have worked with assemblies of over 10,000 parts.

Here's a list of things I've noticed that aren't right:
- The assembler should only work with Parts and Assemblers.
- Each part or assembler should have the option to save internally or externally, as there are assemblers that consist of multiple parts from external files. If changes are made externally, they should reflect when opening a file that uses them. For example, when designing a car, one team works on the chassis and another on the engine. The manager provides the engine mounting points to one team and to another. In the end, an assembler integrates the engine with the car. Within the engine, there may be a need to include an external part and import it into another car, or that part may be part of several cars.
For now, that's all. If you can solve these problems, it's already a good start. 

by the way, check my coment in this pull: https://github.com/FreeCAD/FreeCAD/pull/10764

Thank you to the entire FreeCAD team!

### Full version info

```shell
0.22 dev 36117 release
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-02-26:
Thanks for the report.
For the grounded joint I can see solidworks handling it correctly.
However I do not know how we can achieve this (yet at least). Currently the solver needs a grounded part. Besides the dragging is not very robust yet so removing the requirement to ground is going to be hard.

luisDias89, 2024-02-26:
Thank you. The idea is noted, and I know you'll get there. 
By the way, I would like to contribute with code, but I don't know how to start. . .

PaddleStroke, 2024-02-26:
Do you want to participate on the assembly workbench or on other places ? If you want to contribute that would be great. The first step is to setup a build environment for freecad and setting up git / github : 
https://wiki.freecad.org/Compiling

Then you need to choose one thing you want to do/solve. I suggest starting with something not too ambitious to get started.

tobiasfalk, 2024-02-26:
Could be a "Virtual" Ground, what I mean is that the first part added or used in a joint is used as ground. If a ground is nor already specified

luisDias89, 2024-02-26:
@PaddleStroke I've already tried to install and compile. But I don't know what to do with the libpack. For someone who's just starting, it's confusing. Can't you provide a step-by-step guide or video on how to compile (for beginners)?

PaddleStroke, 2024-02-27:
@tobiasfalk yes we could default to the first inserted part. But it's not like in solidworks where both parts are floating.

@luisDias89 on windows I seem to recall that the ['compile on windows'](https://wiki.freecad.org/Compile_on_Windows) guide is good. For the libpack, you need to first click configure in cmake, then it will fail but it will create entries where you can put the path to the libpack folder.

pierreporte, 2024-08-28:
I agree that it can be practical to define the grounded joint after other operations. There can be an option though to ground the first part.

PaddleStroke, 2025-11-27:
Currently this issue is partly solved. Because you can : 
- Create a joint between 2 ungrounded parts.
- Move 2 ungrounded parts connected by a joint (they move together as a rigid).

The only thing you cannot do, is actionate joints (rotate a part around its revolute by dragging).

So I don't know if this issue can be considered closed? @luisDias89

pierreporte, 2025-11-27:
I agree with @PaddleStroke especially since #24484 doesn't actually exist anymore.

I don't know how we could define the movement of a bunch of joined parts witting an assembly, because it would be needed to define some sort of temporarily grounded part based on some criterion that needs to be defined.

luisDias89, 2025-11-29:
@PaddleStroke This feature is very important to be able to see multiple movements at the same time.
I agree with @pierreporte. A key could be used to activate a mode where moving a part affects only the selected body, without solving all the dependencies. It would also be useful to choose, as a default setting, whether dependencies should be treated as a rigid body or as a hierarchical assembly resolution.

luisDias89, 2025-11-29:
Another thing I’ve noticed is that the move tool is not solving the assembly in real time, and everything becomes temporarily misaligned until I perform a recompute. Maybe my nexts issues for Freecad.
However, I will wait until these features are more developed before continuing my tests.