FreeCAD/FreeCAD #12597: Sketch default point size in 3D view

ofpkubi, 2024-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried to set the default point size of the sketches. In the View tab there is a parameter for point size defaults to 4.
Changing the DefaultShapePointSize parameter (either through preferences or directly in the config file) changes only the point sizes for 3D objects, but not for the sketches.

[image]
[image]

In the source code I found this [src/Mod/Sketcher/Gui/ViewProviderSketch.cpp]:
```

    // Default values that will be overridden by preferences (if existing)
    PointSize.setValue(4);
    
    // visibility automation and other parameters: update parameter and property defaults to follow
    // preferences
    pObserver->initParameters();
    pObserver->subscribeToParameters();
```

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git) AppImage
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-25:
You can set the point size in the Preferences>Sketcher

closed 2024-02-25 by maxwxyz

ofpkubi, 2024-02-25:
> You can set the point size in the Preferences>Sketcher

Thanks!
Something is strange then, because there is no parameter for it.
[image]