FreeCAD/FreeCAD #12599: [Problem] Display of label of the preselection has been lost

azuk, 2024-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A long time ago status bar was modified to display object label as part of the preselection information (https://tracker.freecad.org/view.php?id=3153). Unfortunately later changes have dropped the label part. I would like to request the label to be displayed again. I'm using 0.21 branch version but I checked that the label is not displayed by the main branch version either.

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: recent FreeCAD-0-21 branch, see hash
Build type: Debug
Branch: master
Hash: f191fde3b1d05eed6491a3874a14be7de2d39dc3
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.3.0
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * Render 2023.8.18
  * BIM 2021.12.0
  * Reinforcement
  * Manipulator 1.5.5
  * Woodworking 0.21.1.33694
  * Assembly4.backup1662941442.2770898
  * parts_library
  * Assembly4 0.50.6
  * fasteners 0.4.75
  * Help 1.0.3
  * Silk 0.1.5
  * dodo 1.0.0
  * Curves 0.6.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-25:
https://github.com/FreeCAD/FreeCAD/commit/32122948479893c9c0b4cec6849a5367b07edcef

luzpaz, 2024-03-13:
@wwmayer this is an commit from you back in 2017

wwmayer, 2024-03-14:
Yes, I once added it. 

The object label was removed again with c744157e9a2811
By reading the commit log or code changes there is no reason why it was dropped.

maxwxyz, 2025-10-15:
This should be implemented with the option to hide it, similar to the change to all status bar elements (hints, measurements,...)