FreeCAD/FreeCAD #12601: Path Workbench - EdgeWalker has no edges to load

gagagu, 2024-02-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,
i've created a flat panel from an imported SVG file and add a path job.
I will always get an error on Poket operation on a hole and I don't know why.

> 09:32:28  Part::Refine: Link(s) to object(s) 'Extrude' go out of the allowed scope 'Extrude017'. Instead, the linked object(s) reside within 'Part'.
09:34:42  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\Mod\Path\PathScripts\PathUtils.py", line 70, in new_function
    res = function(*args, **kwargs)
  File "C:\Program Files\FreeCAD 0.21\Mod\Path\Path\Op\Base.py", line 817, in execute
    result = self.opExecute(obj)
  File "C:\Program Files\FreeCAD 0.21\Mod\Path\Path\Op\Area.py", line 366, in opExecute
    for shp in self.areaOpShapes(obj):
  File "C:\Program Files\FreeCAD 0.21\Mod\Path\Path\Op\PocketShape.py", line 126, in areaOpShapes
    self.vWires = [
  File "C:\Program Files\FreeCAD 0.21\Mod\Path\Path\Op\PocketShape.py", line 127, in <listcomp>
    TechDraw.findShapeOutline(shape, 1, FreeCAD.Vector(0, 0, 1))
<class 'ValueError'>: EdgeWalker has no edges to load

09:34:42  Recompute failed!

I've spend hours to get it working but no success. The only way is to export it as *.stl and import it but than all circles are segmented. I've created a "simpe copy","refined copy", delete all jobs and create new, export as step and import it again.

I've created a Post in the Freecad Forum and the Users thinks that it could be a bug and suggested me to create this ticket.

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.60n
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

I've uploaded an Image, the FreeCad file and the ToolBit file to my share [here](https://my.hidrive.com/share/b11w.jtm3x)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-25:
could you try this with the current weekly builds as there were some changes to the PathWB: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-03-13:
Hi @maxwxyz, any idea who's working on CAM wb these days beside sliptonic ?

maxwxyz, 2024-03-13:
no. @djmdjm @davidgilkaufman did a few PRs recently.

davidgilkaufman, 2024-03-13:
I am struggling to reproduce the bug -- I've got some guesses, but if they are not relevant then I need more precise instructions to reproduce. Here's what I've done:

1. Start with the [model attached](https://forum.freecad.org/viewtopic.php?p=656820#p656820) in the forum
2. Note that this model already has errors in it. To avoid confusing these old errors with errors on my changes I deleted Body_in. I also hid Body_out and operations in Job_out to avoid confusion.
3. Following/interpretting [instructions from jescombe](https://forum.freecad.org/viewtopic.php?p=656834#p656834), I copied BaseFeature from Body_out (just BaseFeature -- in the Object Selection/dependencies dialog I unchecked Model_top_mir3) and pasted a copy. Make it visible
3b. Same instructions -- now make a new Path Job from that BaseFeature, with default settings
3c. Same instructions -- create an adaptive operation in that job, select and add the face as Base Geometry, enable checkbox for Use Outline, Ok. We have now recreated the adaptive operation of interest and want to add extensions
3d. Same instructions -- reopen the dialog for that adaptive, Extensions -> Enable Extensions; check the boxes for Edge158 and Edge162.

The instructions here seem to successfully recreate the extensions shown in jescombe's screenshot, which I think is what Siltti is looking for. In this process I did not generate any EdgeWalker errors.

I did some additional messing around and managed to produce "EdgeWalker has no edges to load" errors by deleting geometry items used by operations, but I don't think that is a bug. I'd agree that the error could be a lot clearer, but it is correct for those operations to produce an error in this situation. This is my best guess for how @gagagu gets these errors, but it's hard to be sure.

As an additional test, I tried creating a Pocket operation in this new job to mirror what I think the Iso Kuoppa operation is supposed to do. I added all of the bounded faces and (just to make a clear screen shot) turned the step over down to 10%. It seems to work as I would expect -- screenshot attached

[image]

I'd be willing to look again with more specific reproduction instructions, but barring more details I would close this bug as "cannot reproduce"

luzpaz, 2024-03-13:
@davidgilkaufman thanks for the feedback!

> I'd agree that the error could be a lot clearer

Any suggestions on what that would look like ?

davidgilkaufman, 2024-03-14:
I'd say that whenever we understand what went wrong we should print that explanation, in a message crafted for users to read, instead of printing an exception and backtrace. I'd have to read the code more closely to be sure what's appropriate, but something along the lines of: "Error, base geometry items <list of names> are missing"

We don't do this anywhere else (as far as I know) so I'm a little reluctant to suggest it, but it bothers me that we print error messages to the console without a clear indication of which operation produced them. I would consider adding the operation name to that error message as well.

sliptonic, 2024-04-27:
The operation fails because the selected base geometry is a spline.   Interestingly, the profile operation works just fine with the spline boundary.  This should be a relatively easy solution.

dbtayl, 2024-10-09:
Might be related to this question I had: https://forum.freecad.org/viewtopic.php?p=786201#p786201

kinda looking like when circles/cylinders go through EdgeWalker, they collapse to a single point. EdgeWalker:: makeUniqueVList is where I got to that looked like it might cause that, though I don't know that's where a solution belongs.

luzpaz, 2024-10-10:
Anyone want to take point on this ?

dbtayl, 2024-10-14:
I'll possibly end up looking at it eventually, but I definitely would not count myself as taking point. I'm working through a backlog of half-finished projects before committing to anything else, but still trying to help out where I can.

dbtayl, 2024-10-14:
In AppTechDrawPy.cpp Py::Object findShapeOutline, looks like the issue is if the ONLY edges provided are closed, those get thrown out. The below seems to "fixe" it, but I rather doubt it's a correct fix- I'm not even sure this is the root of the problem. Hopefully somebody who knows better how these functions are supposed to work can propose something better.

See the lines commented "JANKY \"FIX\"" below.

    Py::Object findShapeOutline(const Py::Tuple& args)
    {
        PyObject *pcObjShape(nullptr);
        double scale(1.0);
        PyObject *pcObjDir(nullptr);
        if (!PyArg_ParseTuple(args.ptr(), "OdO", &pcObjShape,
                                                 &scale,
                                                 &pcObjDir)) {
            throw Py::TypeError("expected (shape, scale, direction");
        }

        if (!PyObject_TypeCheck(pcObjShape, &(TopoShapePy::Type))) {
            throw Py::TypeError("expected arg1 to be 'Shape'");
        }

        if (!PyObject_TypeCheck(pcObjDir, &(Base::VectorPy::Type))) {
            throw Py::TypeError("expected arg3 to be 'Vector'");
        }

        TopoShapePy* pShape = static_cast<TopoShapePy*>(pcObjShape);
        if (!pShape) {
            Base::Console().Message("TRACE - AATDP::findShapeOutline - input shape is null\n");
            return Py::None();
        }

        const TopoDS_Shape& shape = pShape->getTopoShapePtr()->getShape();
        Base::Vector3d dir = static_cast<Base::VectorPy*>(pcObjDir)->value();
        std::vector<TopoDS_Edge> edgeList;
        try {
            edgeList = DrawProjectSplit::getEdgesForWalker(shape, scale, dir);
        }
        catch (Standard_Failure& e) {

            throw Py::Exception(Part::PartExceptionOCCError, e.GetMessageString());
        }

        if (edgeList.empty()) {
            return Py::None();
        }

        std::vector<TopoDS_Edge> closedEdges;
        edgeList = DrawProjectSplit::scrubEdges(edgeList, closedEdges);
        
        // JANKY "FIX": if there are closed edges, and no other edges, use those?
        if(edgeList.size() == 0 && closedEdges.size() != 0)
        {
            edgeList = closedEdges;
        }
        // END JANKY "FIX"

        PyObject* outerWire = nullptr;
        std::vector<TopoDS_Wire> sortedWires;
        try {
            EdgeWalker eWalker;
            sortedWires = eWalker.execute(edgeList);
        }
        catch (Base::Exception &e) {
            e.setPyException();
            throw Py::Exception();
        }

        if(sortedWires.empty()) {
            Base::Console().Warning("ATDP::findShapeOutline: Outline wire detection failed\n");
            return Py::None();
        } else {
            outerWire = new TopoShapeWirePy(new TopoShape(*sortedWires.begin()));
        }
        return Py::asObject(outerWire);
    }

dbtayl, 2024-10-14:
This seems to fix what I'm seeing, though it's very likely the wrong fix, and may not even address the root issue. Hopefully somebody who knows how these functions are actually supposed to work can weigh in with a more correct/complete fix.

In AppTechDrawPy.cpp:

    Py::Object findShapeOutline(const Py::Tuple& args)
    {
        PyObject *pcObjShape(nullptr);
        double scale(1.0);
        PyObject *pcObjDir(nullptr);
        if (!PyArg_ParseTuple(args.ptr(), "OdO", &pcObjShape,
                                                 &scale,
                                                 &pcObjDir)) {
            throw Py::TypeError("expected (shape, scale, direction");
        }

        if (!PyObject_TypeCheck(pcObjShape, &(TopoShapePy::Type))) {
            throw Py::TypeError("expected arg1 to be 'Shape'");
        }

        if (!PyObject_TypeCheck(pcObjDir, &(Base::VectorPy::Type))) {
            throw Py::TypeError("expected arg3 to be 'Vector'");
        }

        TopoShapePy* pShape = static_cast<TopoShapePy*>(pcObjShape);
        if (!pShape) {
            Base::Console().Message("TRACE - AATDP::findShapeOutline - input shape is null\n");
            return Py::None();
        }

        const TopoDS_Shape& shape = pShape->getTopoShapePtr()->getShape();
        Base::Vector3d dir = static_cast<Base::VectorPy*>(pcObjDir)->value();
        std::vector<TopoDS_Edge> edgeList;
        try {
            edgeList = DrawProjectSplit::getEdgesForWalker(shape, scale, dir);
        }
        catch (Standard_Failure& e) {

            throw Py::Exception(Part::PartExceptionOCCError, e.GetMessageString());
        }

        if (edgeList.empty()) {
            return Py::None();
        }

        std::vector<TopoDS_Edge> closedEdges;
        edgeList = DrawProjectSplit::scrubEdges(edgeList, closedEdges);
        
        // JANKY "FIX": If no edges are found, but closed edges are, use those?
        if(edgeList.size() == 0 && closedEdges.size() != 0)
        {
			edgeList = closedEdges;
		}
		// END JANKY "FIX"

        PyObject* outerWire = nullptr;
        std::vector<TopoDS_Wire> sortedWires;
        try {
            EdgeWalker eWalker;
            sortedWires = eWalker.execute(edgeList);
        }
        catch (Base::Exception &e) {
            e.setPyException();
            throw Py::Exception();
        }

        if(sortedWires.empty()) {
            Base::Console().Warning("ATDP::findShapeOutline: Outline wire detection failed\n");
            return Py::None();
        } else {
            outerWire = new TopoShapeWirePy(new TopoShape(*sortedWires.begin()));
        }
        return Py::asObject(outerWire);

sslupsky, 2024-11-07:
This is an interesting proposal.  I regularly run into issues where cylinders have two faces when using a polar pattern to replicated them.  When I try to create a tool path, these cylinders are not recognized as "circles" and are ignored by the drilling and helix drilling operations.  You can pick them up as "holes" in the profile operation but that is suboptimal because this operation does not generate a list of base geometries so you cannot edit it to remove or add geometries.

I have noticed that if you manually select the faces of some of these cylinders you can generate a tool path.  But this does not work all the time and there are cases that will not generate a path and you get an index error.

I noticed that if I select the edges of the cylinders that generate the index error instead of the face I can generate a tool path.  I cannot say yet if this work around works in all cases but it is an improvement.  The is not optimal because selecting edges is tedious.  Box Element Selection does not help because it selects the face.  We need a Box Element Selection that allows you to select the type of element you want to select.

So, what you have proposed seems it might help in this situation.

jbaehr, 2025-01-05:
@WandererFan, could you please take a look at what @dbtayl proposed [in the comment above](https://github.com/FreeCAD/FreeCAD/issues/12601#issuecomment-2409648435)? This defect here is about CAM, but the actual error comes from the TechDraw method `findShapeOutline` that's used here.
It's quite easy to reproduce:
```py
>>> r = Part.makePolygon([(0,0,0), (1,0,0), (1,1,0), (0,1,0), (0,0,0)])
>>> c = Part.makeCircle(1)
>>> TechDraw.findShapeOutline(r, 1, FreeCAD.Vector(0, 0, 1))
<Wire object at 0x6000011a8dc0>
>>> TechDraw.findShapeOutline(c, 1, FreeCAD.Vector(0, 0, 1))
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ValueError: EdgeWalker has no edges to load
```

dbtayl, 2025-01-07:
This also causes more subtle issues in finding the boundary if the true boundary is a closed single-edge wire (eg, circle), but it has a cutout in the middle. In the screenshot below, the small rectangular cutout is detected as the "outline".
[image]

My janky fix above doesn't catch that case- replacing it with `edgeList.insert( edgeList.end(), closedEdges.begin(), closedEdges.end() );` (ie, just joining the two vectors/lists) seems to work in trivial testing, but I'm still not super confident it's a "correct" fix.

dbtayl, 2025-01-10:
Looks like `findOuterWire` has the same issue. I haven't dug through the other functions, but `edgeWalker` uses the same `DrawProjectSplit::scrubEdges` call that appears to be causing issues elsewhere.

closed 2025-01-27 by yorikvanhavre