FreeCAD/FreeCAD #12604: [Problem] Macros = ModuleNotFoundError - No module names gui_facebinders

RandomLegend, 2024-02-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I wanted to record a macro that creates a facebinder via the draft workbench on a selected face. Using either the gui buttons or using shortcuts, i can create the facebinder without issues.

Executing the macro however throws the following error:

```
12:55:12  pyException: Traceback (most recent call last):
  File "/home/frank-garuda/.local/share/Ondsel/Macro/Facebind.FCMacro", line 6, in <module>
    import gui_facebinders
  File "/tmp/.mount_Ondsell3G3CT/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'gui_facebinders'
```

Tested this with the weekly build i currently use, the stable version and the repackaged Ondsel version. All behave the same.

### Full version info

```shell
OS: Garuda Linux (Hyprland/garuda-hyprland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35594 (Git) AppImage
Build type: Release
Branch: main
Hash: 49e7bba9c7390b11e6cc9acebd10cc1568ec8869
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-02-26:
Can you provide the macro for testing purposes ?

RandomLegend, 2024-02-26:
```
# -*- coding: utf-8 -*-

# Macro Begin: /home/frank-garuda/.local/share/FreeCAD/Macro/Test.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++
import FreeCAD
import PartDesignGui
import gui_facebinders
import Draft

# Gui.runCommand('Std_DlgMacroRecord',0)
# Gui.Selection.addSelection('Unnamed','Body','Pad.Face6',21.6569,8.72649,10)
### Begin command Std_Workbench
# Gui.activateWorkbench("DraftWorkbench")
### End command Std_Workbench
### Begin command Draft_Facebinder
sels = FreeCADGui.Selection.getSelectionEx('', 0)
facebinder = Draft.make_facebinder(sels)
Draft.autogroup(facebinder)
FreeCAD.ActiveDocument.recompute()
### End command Draft_Facebinder
# Gui.Selection.addSelection('Unnamed','Facebinder')
# Macro End: /home/frank-garuda/.local/share/FreeCAD/Macro/Test.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++

```

This was generated via Macro Recording and resulted in this error:

```
15:51:31  pyException: Traceback (most recent call last):
  File "/home/frank-garuda/.local/share/FreeCAD/Macro/Test.FCMacro", line 6, in <module>
    import gui_facebinders
  File "/tmp/.mount_FreeCAwwOnnt/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'gui_facebinders'

```

___

Deleting facebinder_gui in the import section resolves the issue and executes the macro as intended.

wwmayer, 2024-02-27:
The correct import statement should be: `from draftguitools import gui_facebinders` But I have no idea where the line `import gui_facebinders`. 

Since you use a fork of FreeCAD the problem may come from this fork.

luzpaz, 2024-03-06:
@RandomLegend please respond.

RandomLegend, 2024-03-06:
Sorry guys, had much going on

So i fixed it:

This is the correct script now:

```
import FreeCAD
import PartDesignGui
import Draft
import Part
import PartDesign

# Gui.runCommand('Std_DlgMacroRecord',0)
### Begin command Part_Loft
from FreeCAD import Base

# Get document name
doc_name = App.activeDocument().Label

# Gui.runCommand('Std_DlgMacroRecord',0)
### Begin command Std_Workbench
# Gui.activateWorkbench("DraftWorkbench")
### End command Std_Workbench
### Begin command Draft_Facebinder
sels = FreeCADGui.Selection.getSelectionEx('', 0)
facebinder1 = Draft.make_facebinder(sels[0])
facebinder1.Label = "Facebinder_Loft_1"
facebinder2 = Draft.make_facebinder(sels[1])
facebinder2.Label = "Facebinder_Loft_2"
Draft.autogroup(facebinder1)
Draft.autogroup(facebinder2)
FreeCAD.ActiveDocument.recompute()
App.getDocument(doc_name).addObject('Part::Loft','Loft')
App.getDocument(doc_name).ActiveObject.Sections=[facebinder1, facebinder2, ]
App.getDocument(doc_name).ActiveObject.Solid=True
App.getDocument(doc_name).ActiveObject.Ruled=False
App.getDocument(doc_name).ActiveObject.Closed=False
```

So unlike @wwmayer proposed it wasn't needed to do from `draftguitools import gui_facebinders`. The `import gui_facebinders` came from the Macro recording.

Also @wwmayer yes i posted the results from the fork (ondsel) but i can replicate this macro identically on the original freecad version.

wwmayer, 2024-03-06:
> but i can replicate this macro identically on the original freecad version.

How?

RandomLegend, 2024-03-06:
1. Start Macro Record
2. Select face of body
3. Switch to "Draft" Workspace
4. Create facebinder on the selected face
5. Stop Macro Recording

It created the import gui_facebinders on the stable version and the 0.22.dev version that i also had installed.

wwmayer, 2024-03-06:
Confirmed! Thanks.

I will reopen the bug report because recorded macro code must be correct.

wwmayer, 2024-03-06:
The function Application::sAddCommand needs be fixed to determine the module and package names.
E.g. for the registered face binder command it determines the module name "gui_facebinders" without the package name "draftguitools" and passes it to the PythonCommand/PythonGroupCommand with setAppModuleName.

This app module name is written to the recorded macro.

luzpaz, 2024-08-30:
Can this ticket now be closed @wwmayer ?

maxwxyz, 2025-11-01:
There is a fix for this issue https://codeberg.org/wwmayer/FreeCAD11/commit/bd279c38a76bd7cf55df5e5168d6addddd22586d

maxwxyz, 2026-11-07:
There is another fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/d49d51915d9ff143943ca54b272d77d4c9aaeb69

Recorded commands use the importable package name, so Facebinder no longer records import gui_facebinders on its own. This also keeps the filename fallback for ordinary scripts.