FreeCAD/FreeCAD #12605: [Problem] CarbonCopy not working between bodies, depends on visibility automation properties

chrisb-github, 2024-02-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

CarbonCopy doesn't work between different bodies, if the sketch to be copied has the visibility automation properties set as follows:
ForceOrtho=false
HideDependent=true
RestoreCamera=true
SectionView=true
ShowLinks=false
ShowSupport=true

it works, when
ForceOrtho=false
HideDependent=false
RestoreCamera=false
SectionView=true
ShowLinks=false
ShowSupport=false

It works also in the latter case when SectionView=false

### Full version info

```shell
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.2.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fasteners 0.4.75
  * sheetmetal 0.3.14
  * ExplodedAssembly
  * Curves 0.6.21
```
```

### Subproject(s) affected?

Sketcher

### Anything else?

Corresponding forum discussion: https://forum.freecad.org/viewtopic.php?t=85515
To reproduce:
- rename the extension to .fcstd
- edit Sketch001
- with Ctrl pressed (CMD on macOS) hover over the other sketches.
- see that you can copy sketch CopyFrom_cb (thick lines)
- see that you cannot copy Sketch (thin lines)
[carboncopy_cb.zip](https://github.com/FreeCAD/FreeCAD/files/14396848/carboncopy_cb.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gryphonB-2, 2024-11-23:
Just to confirm this bug.
I think it's clear that, by whatever mechanism, having a copy of a sketch in multiple bodies can be very useful.
I am currently working on a Part that would benefit from copies in three Bodies. 
Ability to use a CarbonCopy would allow me to create a generic Part, inserting different cross sections into a master sketch, and having the required bodies update automatically - ie a parametric flow based on a single sketch.
###
As an aside/Feature Request, it would be very useful to be able to Clone a sketch, at present not available for 2D objects.

FEA-eng, 2025-06-22:
I wanted to test it but, as mentioned in #12606, those properties cannot be changed.

FEA-eng, 2026-01-22:
See [this](https://github.com/FreeCAD/FreeCAD/issues/12605#issuecomment-3216789983)

FEA-eng, 2026-10-23:
See [this](https://github.com/FreeCAD/FreeCAD/issues/12605#issuecomment-3437938001)