FreeCAD/FreeCAD #12615: Grouping of Parts in the Assembly

tobiasfalk, 2024-02-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be nice if one could create Groups/Folders inside the Assembly so that one could group and organize parts.
One common thing that would be nice to group are Fasteners, so that they do not clouder the Tree View too much.
Yes one will say why not do just Sub Assemblies when they are available but even then it would be nice for things like Fasteners and co.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36082 (Git)
Build type: Release
Branch: main
Hash: 7a025e4e60a607bdbdd1599627ce13bb8eeda071
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.56a
  * AirPlaneDesign 0.4.0
  * Assembly3 0.12.0
  * Assembly4 0.11.10
  * CADExchanger
  * cadquery_module
  * CurvedShapes 1.0.5
  * dxf-library
  * DynamicData 2.46.0
  * ExplodedAssembly
  * fasteners 0.5.2
  * fcgear 1.0.0
  * FeedsAndSpeeds 0.5.0
  * FEM_FrontISTR 0.1.0
  * FreeCAD_Electric
  * FreeCAD_Schematics
  * GDML 2.0.0
  * kicadStepUpMod 10.11.7
  * Plot 2022.4.17
  * pyrate
  * Render 2022.2.0
  * sheetmetal 0.2.56
  * Ship 2022.4.11
  * SteelColumn
  * ThreadProfile 1.82.0
  * timber
  * trails 2022.1.0
```
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-26:
Related: https://github.com/FreeCAD/FreeCAD/issues/12142

tobiasfalk, 2024-02-26:
> Related: https://github.com/FreeCAD/FreeCAD/issues/12142

Sorry didn't find this in my search🤦‍♂️

PaddleStroke, 2024-02-26:
You can already use group inside assembly : 
[image]
Did you find cases where it was not working as expected? If no then this should be closed. If yes please give me a sample file to debug it.

tobiasfalk, 2024-02-27:
I do not know if I am blined but I do not find it, I even downloaded the newest Weekly Build

tobiasfalk, 2024-02-28:
Ok, found it. For some reason Structure Toolbar was disabled

tobiasfalk, 2024-02-28:
But I do get This dialog for every part i move in to a group, a checkbox to select "do for all" would be nice.
[image]

PaddleStroke, 2024-02-28:
This message should not appear in this case. I'll work on it

maxwxyz, 2024-07-05:
@PaddleStroke I think it is the same problem as https://github.com/FreeCAD/FreeCAD/issues/15111 but cannot confirm.

PaddleStroke, 2024-07-05:
No it's not the same. This was about grouping parts. Which is actually working. But there was still the small issue that when you moved a part to a group, it would prompt you to delete the joints. Which should not be mandatory as the object stays in the assembly.

Anyway this is going to be fix by https://github.com/FreeCAD/FreeCAD/pull/15012

closed 2024-07-15 by yorikvanhavre