FreeCAD/FreeCAD #12620: [Problem] UI/UX: Tree view overlapping 3D view

maxwxyz, 2024-02-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As mentioned in the FreeCAD Day 2024 Complaint Session:
 **Tree view should not get in the way of the rest of the UI.**
Based on that, tree view should be somewhere alone, transparent on the side, on the screen somewhere.
Tree view should not get in the way of the rest of the UI.

Maybe this could be changed in the default state @MisterMakerNL Currently, all dialogs are at the left side, from top to bottom:
1. Task Dialog
2. Tree View
3. Property View

Maybe Task Dialog could be moved to the right, Tree View could stay on the left and Property either below the tree or hidden, as https://github.com/FreeCAD/FreeCAD/issues/11072 was implemented. 
Maybe even the transparent overlays should be enabled by default.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-02-26:
Transparent overlay for the three view by default is IMO a very good idea

yorikvanhavre, 2024-02-26:
For me we're mixing two things here... When we say "tree view should not get in the way of the UI" I feel like saying "why?"

Now, if we say "tree view should not get in the way of the model" then I agree. And then it's IMHO best to keep it in a separate window than printing transparently on top of the model. That's the basic principle of a non-overlapping interface, like Blender for example.

Now those are two different directions (overlapping or non-overlapping), and there are people who prefer one over the other. If one or the other is the default is IMHO not very important as long as users can switch to the other. But in any case, I fail to see where we have a "problem" here?

Another point to take into consideration, on small screens (13' laptops, for ex) having both left and right dock areas is impractical, you have virtually no 3D view space left :)

MisterMakerNL, 2024-02-26:
Properties view should get a silent dead, it should be a dev tool mostly. 
We might want to work towards a dev mode, where properties view and all the consoles are available. And a casual view where all those are gone. This would simplify the ui a lot, since you won't see all those options anymore. And it would be just one witch to turn on your powertools.
But we aren't ready for this. 
But we should have a focus to make it happen in my opinion.
Properties view is still accessible by right click in casual.
Treeview and task should never overlap, in Solidworks when treeview and taskview overlap a second treeview pops up in overlay mode. But this solution is nice if you have a limited amount of features. And we have an unlimited amount, so I don't see this doable in Freecad.
It might a be a good idea to prevent treeview and taskdialog to merge in each other in the non-dev gui as it has no use there and would only confuse someone or make it difficult to setup properly.

We currently only have the dev-mode. @yorikvanhavre would probably just work in dev mode and ignore the casual mode. Which is totally fine off-course he is a dev. We can also call it Yorik-mode.

yorikvanhavre, 2024-02-26:
I don't understand well how you work with a parametric modeler without seeing the parameters :) But of course if there are enough people interested in working that way, why not... I'm never opposed to changing the defaults, as long as people can change back.

All these panels can be docked anywhere, stacked, turned on and turned off. For me that's the main thing, it needs to be customizable. One user is not another.

pierreporte, 2024-02-26:
> I don't understand well how you work with a parametric modeler without seeing the parameters :)

I think they would be set by the task panel only, like other program.

obelisk79, 2024-02-27:
I'm commenting so I can follow the conversation. Right now, I am working a few other lines of effort so I can't invest in formulating a good proposal for default layout. Although, with the pending advent of an actual quick setting/first run page being added to the new start workbench, I think the existing defaults should remain, and the option for a 'classic' vs 'modern' layout could be applied leveraging the functionality of preference packs and make switching to what is comfortable for users both new and experienced simpler. Also makes layout experimentation safer as a user could always get back to the quick setting page and apply the default of choice as a form of reset.

MisterMakerNL, 2024-02-27:
> > I think they would be set by the task panel only, like other program.
> 
> But then how do you change the parameters of several objects at once ;)

[image]

pierreporte, 2024-02-27:
TBH I didn’t ask for removing the properties entirely, just speculated on the intent. If I got to decide, it would still exist (as a panel like the task panel—a feature request will be opened for this at some point) but it wouldn’t contain design information.

adrianinsaval, 2024-03-02:
> For me we're mixing two things here... When we say "tree view should not get in the way of the UI" I feel like saying "why?"

Cause it's a PITA switching back and forth between task panel tab and tree+property tab, if we can have at least two available at the same time without obstructing the 3d view that's the sweet spot IMO. Transparent tree view is very useful for this, the impact on available 3d view is very minimal. And then you can have the task or property view in ful display at the same time.

Of course, we should absolutely retain the possibility of customizing panels.

adrianinsaval, 2024-03-02:
for what is worth, I love the property view. I've always considered it one of FreeCAD's selling points :) but I always used it separated from the tree, when they are together they both end up annoyingly short and constantly need scrolling.

MisterMakerNL, 2024-03-02:
The sheet metal wb that is been worked on to get an UI, is a great example how we should make the properties view to be moved to the background. The wb now uses the properties view to set all parameters. With a good UI the properties view isn't necessary anymore.
Every CAD program has a properties view so I wouldn't say it's a selling point, only difference with other CAD programs is that in Freecad it is used for normal features and in other CAD programs to do some advance parameter stuff.  I personally don't want to go digging in a properties table to change a flange height.

obelisk79, 2024-03-02:
> for what is worth, I love the property view. I've always considered it one of FreeCAD's selling points :) but I always used it separated from the tree, when they are together they both end up annoyingly short and constantly need scrolling.

I wonder if the property panel would be better served as something which is easily invoked as a floating panel via right click menus (tree or otherwise) so it can be displayed when wanted/needed and safely closed/hidden from view when not. I use that panel frequently too, but not so much that I need it to be ever-present.

MisterMakerNL, 2024-03-02:
@obelisk79 You might want to try the right click option ;)

yorikvanhavre, 2024-03-04:
> Cause it's a PITA switching back and forth between task panel tab and tree+property tab, if we can have at least two available at the same time without obstructing the 3d view that's the sweet spot IMO

Yes I agree with that. Then we can say the problem is much more a conflict between the tree/properties and the task panel, rather than the properties view being too much, although I understand that for some people like @MisterMakerNL it is of no use.

> Transparent tree view is very useful for this, the impact on available 3d view is very minimal

I don't agree there :sweat_smile: For me it's the worse possible thing, when things get complex you don't see neither your model nor your tree view clearly... (Of course that's only my opinion...)

But anyway, then it seems to me the user should be able to decide, right? Between 1) a permanent properties window, 2) a hidden properties view and 3) a "on-demand" properties view. What about that?

Basically, it's the task panel that is now kind of on-demand. The idea there would be to allow the opposite...

yorikvanhavre, 2024-03-04:
This works on big screens, less on smaller ones. For me the workflow is way better and clearer when separated in non-overlapping windows:

[image]

[image]

But of course this is much a matter of personal preferences. As long as both are available, everybody should be happy :)

Remains the issue of task panel vs tree view and the annoying switch. Again, when you have a big screen, nothing is really a problem. The problem is when space is limited.

MisterMakerNL, 2024-03-04:
fyi Light theme is currently broken with overlay.

luzpaz, 2025-08-21:
Is this still relevant ?

MisterMakerNL, 2025-08-22:
I actually tried putting the background on the tree items but couldn't get it to apply. Opentheme currently breaks active and edited background state.