FreeCAD/FreeCAD #12625: When using TechDraw the anchor view of the projection group is greyed out and views are shifted sideways in the projecton ui.

gazzatav, 2024-02-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a body in part design.
2. Pad a sketch - I just did a cylinder or box with same result.
3. Open TechDraw workbench and open a template or default page.
4. Insert Projection Group.

At this point the anchor view is greyed out but the (front) view that would have been there is one box to the right (with the tool tip and frame label saying "Front".  See image below.

Other possibly related problem:
Double click on project group in tree with a different drawing visible and FreeCad sometimes crashes with segmentation fault.

### Full version info

```shell
OS: Ubuntu Core 22 (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36128 (Git) Snap 937
Build type: Release
Branch: main
Hash: a38e73135e95b1f1845f1b390b823bf8b386d026
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.50.7
  * sheetmetal 0.4.1
```
```

### Subproject(s) affected?

Techdraw

### Anything else?

Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'TypeError'>: wrong type for adding projection
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: wrong type for adding projection
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'TypeError'>: wrong type for adding projection
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: wrong type for adding projection

![image](https://github.com/FreeCAD/FreeCAD/assets/14364360/ca295532-749b-49a6-a6c2-74e4548756ab)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gazzatav, 2024-02-26:
[image]

gazzatav, 2024-02-26:
[image]

Be-Cool-Man, 2024-02-26:
Can confirm with

```
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36117 (Git) AppImage
Build type: Release
Branch: main
Hash: 8a3407aee6dff8218282d9cdc3f4e32e7ad958d4
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE) [ OS: German/Germany (de_DE) ]
Installed mods: 
  * BillOfMaterials 0.0.5.6
  * Assembly4 0.50.7
  * A2plus 0.4.63
  * BOLTSFC
  * ExplodedAssembly
  * Assembly3 0.12.2
  * sheetmetal 0.4.2
  * dodo 1.0.0
  * fcgear 1.0.0
  * Curves 0.6.23
  * OpenDark 2023.12.17
  * fasteners
  * Help 1.0.3
```
---
However, when I click on one of the empty boxes, I get the following error and the box doesn't get the check mark until I move the cursor out of the box. Additionally, the corresponding view will NOT be added.
```
22:12:53  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'TypeError'>: wrong type for adding projection
22:12:53  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: wrong type for adding projection
```

gazzatav, 2024-02-28:
I have created a thread in the forum https://forum.freecad.org/viewtopic.php?p=743599#p743599 which has some more detail and images.  The problem seems to have worsened between the weekly and edge version this week.  I also see the error message above in edge.  The projection group in the TechDraw WB has become somewhat unusable and crashes are more regular.  Hopefully this trace will yield some useful information:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'RuntimeError'>: Project Group missing Anchor projection item
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Project Group missing Anchor projection item
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f01292f8520]
#1  0x7f004ccb38f7 in TechDrawGui::TaskProjGroup::saveGroupState() from /snap/freecad_edge/940/usr/lib/TechDrawGui.so+0x1b7
#2  0x7f004ccb46f3 in TechDrawGui::TaskProjGroup::TaskProjGroup(TechDraw::DrawProjGroup*, bool) from /snap/freecad_edge/940/usr/lib/TechDrawGui.so+0xba3
#3  0x7f004ccb4b33 in TechDrawGui::TaskDlgProjGroup::TaskDlgProjGroup(TechDraw::DrawProjGroup*, bool) from /snap/freecad_edge/940/usr/lib/TechDrawGui.so+0x63
#4  0x7f004ce7b507 in TechDrawGui::ViewProviderProjGroup::setEdit(int) from /snap/freecad_edge/940/usr/lib/TechDrawGui.so+0x77
#5  0x7f004ce7b253 in TechDrawGui::ViewProviderProjGroup::doubleClicked() from /snap/freecad_edge/940/usr/lib/TechDrawGui.so+0x13
#6  0x7f012c392f15 in Gui::TreeWidget::mouseDoubleClickEvent(QMouseEvent*) from /snap/freecad/940/usr/lib/libFreeCADGui.so+0x3a5
#7  0x7f012a5e82d9 in QWidget::event(QEvent*) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xb99
#8  0x7f012a695ce2 in QFrame::event(QEvent*) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x22
#9  0x7f0129a0abaa in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0xaa
#10  0x7f012a5a4752 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x72
#11  0x7f012a5ac3a4 in QApplication::notify(QObject*, QEvent*) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#12  0x7f012c0fd2e8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/940/usr/lib/libFreeCADGui.so+0x98
#13  0x7f0129a0ae4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#14  0x7f012a5aae87 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#15  /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c91cb) [0x7f012a6011cb]
#16  /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc525) [0x7f012a604525]
#17  0x7f012a5a4763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#18  0x7f012c0fd2e8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/940/usr/lib/libFreeCADGui.so+0x98
#19  0x7f0129a0ae4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#20  0x7f0129e333b8 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x858
#21  0x7f0129e06dbc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#22  /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5(+0x76f5e) [0x7f012376af5e]
#23  /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7f01274b4d3b]
#24  /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x7f012750a258]
#25  /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f01274b23e3]
#26  0x7f0129a64338 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#27  0x7f0129a0976b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#28  0x7f0129a11d14 in QCoreApplication::exec() from /snap/freecad/940/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#29  0x7f012c0386f4 in Gui::Application::runApplication() from /snap/freecad/940/usr/lib/libFreeCADGui.so+0x6c4
#30  /snap/freecad_edge/940/usr/bin/FreeCAD(+0xa709) [0x5607c1385709]
#31  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f01292dfd90]
#32  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f01292dfe40]
#33  /snap/freecad_edge/940/usr/bin/FreeCAD(+0xaac5) [0x5607c1385ac5]
```

WandererFan, 2024-02-29:
Do you need to do anything special to set up the crash?  Either it is fixed along with the view positioning, or I'm being inept.

blinking-led, 2024-03-01:
Add some information about the same bug.

```
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36166 (Git)
Build type: Release
Branch: main
Hash: a5ea8023a29b6d7bbf1ebd57b5ed31dd27c5947b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

When I'm trying to add some projection. As you can see, no new projections appear.

[image]

File from the official website
[R5CQE2085.zip](https://github.com/FreeCAD/FreeCAD/files/14458536/R5CQE2085.zip)

Text message in the log:

> 11:05:07  pyException: Traceback (most recent call last):
>   File "<string>", line 1, in <module>
> <class 'TypeError'>: wrong type for adding projection
> 11:05:07  Unhandled Base::Exception caught in GUIApplication::notify.
> The error message is: wrong type for adding projection
> 11:05:08  pyException: Traceback (most recent call last):
>   File "<string>", line 1, in <module>
> <class 'TypeError'>: wrong type for adding projection
> 11:05:08  Unhandled Base::Exception caught in GUIApplication::notify.
> The error message is: wrong type for adding projection

gazzatav, 2024-03-02:
```
Version: 0.22.0dev.36228 (Git) Snap 943

```
The issue is not longer present in the above revision which I tested on the edge channel.  Thanks, look forward to seeing this fix in the weekly beta channel.

closed 2024-03-02 by gazzatav