FreeCAD/FreeCAD #12638: [Problem] Certain critical functionality is only available in add-ons not core.

maxwxyz, 2024-02-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**Certain critical functionality is only available in add-ons not core.**

Discusion:
Certain critical functionality is only available as external workbenches and is not available by default.
Properties are hidden in an add-on with a non obvious name called FreeCAD Info

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-27:
@sliptonic could you add or link to your survey results? I remember seeing some info on which addons are most used and crucial for many users.

sliptonic, 2024-02-27:
We're publishing the results in pieces.  Part 1 just went up.  I haven't even boosted it yet.  This part deals with demographics.  Results about add-ons is next.

https://ondsel.com/blog/freecad-user-survey-results-part-1/

maxwxyz, 2024-03-08:
Here are the results:
[image]
Source: https://ondsel.com/blog/freecad-user-survey-results-part-2/

FEA-eng, 2024-03-12:
The fact that the Surface workbench is included in this poll must be a mistake since it's already built-in. But it's good to see that the most commonly used (and thus having the biggest potential for integration) add-ons are the ones for crucial CAD stuff like fastener libraries and sheet metal modeling. IMO the Curves wb has a huge potential for integration with the Surface wb but it's totally up to its developer.

Macros are much easier to integrate so it would be good to know which ones are the most popular too. Some cover essential CAD functionality (like FCInfo which is not a workbench btw. or parametric curve).

All in all, I always repeat on the forum that I like the modularity of FreeCAD and the availability of add-ons for even really niche applications but I definitely agree that some features available only in add-ons should be built-in for user-friendliness, availability, ease of maintenance and so on.

FlachyJoe, 2024-03-12:
> I definitely agree that some features available only in add-ons should be built-in for user-friendliness, availability, ease of maintenance and so on.

I don't because it's great to have separate code repositories to sort issues and PRs. And I'd like to keep a small package.

FEA-eng, 2024-03-12:
>  I don't because it's great to have separate code repositories to sort issues and PRs. And I'd like to keep a small package.

The problem is that those repositories are totally dependent on one person. Package size is also important but often doesn’t change that much (especially when macros are integrated). But again, I’m talking only about the most crucial functionalities that IMO should be available right away in every complete CAD application. I know that many FreeCAD veterans don’t share this view as opposed to most people coming from other software.

FlachyJoe, 2024-03-12:
> The problem is that those repositories are totally dependent on one person.

They can be forked to reside on https://github.com/FreeCAD and we can ask their authors to PR in these ones.

chennes, 2024-04-15:
What would it look like to close this issue? How will we decide when it is "done"?

pierreporte, 2024-04-15:
@chennes I think that we should first list what the critical features are, depending on the category (Mechanical CAD, Architecture CAD, CAM, FEM). Then, when those are all implemented (as a minimum viable product), it could be considered as resolved.

maxwxyz, 2024-08-28:
@pierreporte could CWG define the critical features? We could add separate issues then for the most critical ones and close this. There will always be some features available in addons and not in main.

pierreporte, 2024-08-28:
@maxwxyz That’s why the P1 category is for. When the issues are all triaged, we will have a good idea of the features still lacking in the tracker.

maxwxyz, 2024-08-28:
but most of the issues still need to be created. Who is screening the functions of these addons and adding issues for critical functionality?