FreeCAD/FreeCAD #12644: [Problem] Default Spaceball/mouse/pilot settings flips axes in comaprison to other CAD / trainer program

Kagee, 2024-02-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have always been annoyed on Freecad for it's (apparently) awful 3dconnexion support, even after following the "Known issue"-walkthrough on the wiki. 

Today i discovered that what had always angered me was that the Y and Z inputs were swapped from the normal, and i had to use the "Flip YZ" option, and the mouse was perfectly usable. 

"Normal" here is defined as "How the axes map in other CAD software", *including* in the 3dconnexion trainer software.

Why does it appear that the default option in Freecad (non-swapped YZ) makes the inputs unnatural/wrong compared to how the 3dconnexion devices normally works?

Should this be changed? 

My input device is a 3dconnexion SpaceMouse Wireless. 
I have tested this on fresh installations of Freecad on both Windows (0.22-dev) and Ubuntu (0.21.2)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Norwegian Bokmal/Norway (nb_NO)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Kagee, 2024-02-27:
https://github.com/FreeCAD/FreeCAD/issues/9543 is related, still relevant, but not the same bug/problem.

glabifrons, 2024-06-08:
I agree, this is frustrating. It took me a bit to figure it out too (back around 0.16 +/- .01).
The configuration can be changed in two places, the menu you mentioned and also in /etc/spnavrc, but you are correct that it acts differently in different software.

It seems as though FreeCAD's YZ inversion is itself inverted (since the invert selection must be active for it to behave normally).

FWIW (to provide the perspective of someone who is new to FreeCAD):
I'm trying to introduce a friend to FreeCAD and bought him a SpaceMouse, and he ran into the same problem. Unfortunately, he's in another state so I have to describe the workaround to him. He always says he'll dig it out and try it again, then forgets the workaround by the time he does and gets frustrated again.