FreeCAD/FreeCAD #12645: [Problem] Deleting multiple fillets crashes FreeCAD

chrisb-github, 2024-02-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Deleting in the attached file the last fillets, Fillet016-Fillet034 all at once makes FreeCAD crash. Deleting them one by one from the works well.

To reproduce:
- change the extension of the attached file from zip to fcstd, open file
- select all fillets from Fillet016 to Fillet034
- Delete
- 
->FreeCAD terminates immediately

### Full version info

```shell
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.2.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fasteners 0.4.75
  * sheetmetal 0.4.2
  * ExplodedAssembly
```
```

### Subproject(s) affected?

PartDesign

### Anything else?

The file is from this forum post: https://forum.freecad.org/viewtopic.php?p=743363#p743363
[BuckBoxLidRework.zip](https://github.com/FreeCAD/FreeCAD/files/14426026/BuckBoxLidRework.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chrisb-github, 2024-02-27:
The same happens when selecting multiple chamfers in the same file.

maxwxyz, 2024-02-28:
Is it also happening if you select from bottom to top?

xtemp09, 2024-02-28:
Not reproducible on Linux and Windows. I noticed that there are many issues reporting crash on MacOS, while they are not reproducible on other platforms. =]

iAmVaeVictus, 2024-02-29:
Similar behavior when adding and removing fillets to edges that are not valid (not sure if the invalid part is the trigger or not). 

I cannot give an exact step by step to reliably reproduce like the OP above, but it has happened 3x while trying to fillet the same part. 

RTF file of 2x Apple Crash Reports. 
[FreeCAD 0.22.0 Crash.rtf.zip](https://github.com/FreeCAD/FreeCAD/files/14451438/FreeCAD.0.22.0.Crash.rtf.zip)

```
OS: macOS 14.3
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36117 (Git)
Build type: Release
Branch: main
Hash: 8a3407aee6dff8218282d9cdc3f4e32e7ad958d4
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

xtemp09, 2024-03-01:
@iAmVaeVictus, thank you for providing the crash reports. They are completely useless without the reproduction steps.

chrisb-github, 2024-03-02:
> Is it also happening if you select from bottom to top?

It also happens, if I select the last and then with shift the first in the list.
It doesn't happen if I select them one by one from bottom to top.

closed 2025-06-29 by github-actions[bot]