FreeCAD/FreeCAD #12653: [Problem] Sketcher: endpoint-to-endpoint tangent with the same B-Spline does not work

AjinkyaDahale, 2024-02-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To reproduce:
1. Create a B-spline long enough that the end points are close to each other.
2. Select both endpoints and apply tangent constraint.

The constraint behaves simply as if it were a coincidence constraint.

### Full version info

```shell
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36176 +4 (Git)
Build type: Release
Branch: sketcher-join-curves-2
Hash: a292bb447bc532af8068624791087503ebdba89e
Python 3.11.7, Qt 5.15.12, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.5.12
  * Curves.backup1662220878.4365516 0.4.4
  * ExtremeProDark 2.5.3
  * ProDarkThemePreferencePack 1.0.0
  * ProDarkThemePreferencePack.backup1662220881.5889003 1.0.0
```
```

### Subproject(s) affected?

Sketcher

### Anything else?

The problem is that the same normal is used in constraint solving, since the points are coincident. Further, there is also a "reduction" step in planegcs solver that replaces the variables associated with the points' coordinates with a single variable (since there is an equality constraint), so there is no way of knowing which normal to use even with pointers.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2025-06-22:
Still relevant and even led to "Solver failed to converge" error:

[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-10-06:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```