FreeCAD/FreeCAD #12655: [Problem] [Regression] Intra-Sketch Constraint Failure

frank-trampe, 2024-02-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When attempting to migrate a FreeCAD project from 0.21.2 to a recent weekly build, I am unable to import several sketches. In particular, FreeCAD rejects expressions that reference reference constraints within the same sketch.

```
20:56:39  3.8e-08 <App> Document.cpp(2146): Failed to restore Base_Framing_2_29#Sketch188.ExpressionEngine: Reference constraint from this sketch cannot be used in this expression.
20:56:39  0.0148102 <App> Document.cpp(2146): Failed to restore Base_Framing_2_29#Sketch249.ExpressionEngine: Reference constraint from this sketch cannot be used in this expression.
20:56:39  0.0615039 <App> Document.cpp(2146): Failed to restore Base_Framing_2_29#Sketch294.ExpressionEngine: Reference constraint from this sketch cannot be used in this expression.
20:56:39  0.200217 <App> Document.cpp(2146): Failed to restore Base_Framing_2_29#Sketch475.ExpressionEngine: Reference constraint from this sketch cannot be used in this expression.
20:56:39  0.201672 <App> Document.cpp(2146): Failed to restore Base_Framing_2_29#Sketch480.ExpressionEngine: Reference constraint from this sketch cannot be used in this expression.
20:56:39  0.243651 <App> Document.cpp(2146): Failed to restore Base_Framing_2_29#Sketch556.ExpressionEngine: Reference constraint from this sketch cannot be used in this expression.
```

[Base_Framing_2_29_Package_1.zip](https://github.com/FreeCAD/FreeCAD/files/14385603/Base_Framing_2_29_Package_1.zip)

These expressions seem to work properly in 0.21.2, and `git blame` does not show any significant changes since the release of 0.21.2 in the neighborhood of [the language in the error message.]( https://github.com/FreeCAD/FreeCAD/blob/97f9c4e467cba8e6d855abce2ca5499f3b7dc258/src/Mod/Sketcher/App/SketchObject.cpp#L8633 ).

#10482 mentions that circular references in sketch expressions are undetected as of September 1st of last year, and #11716 purports to fix that, but it does not touch the code in question. Is it possible that there was a rebase or a force push that changed this behavior?

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36082 (Git) AppImage
Build type: Release
Branch: main
Hash: 7a025e4e60a607bdbdd1599627ce13bb8eeda071
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-28:
@bgbsww @mfro0 ping

bgbsww, 2024-02-28:
Looking just at the first error, Sketch188 is "C 3x5 Real" and it has a Flange_Slope_X that is defined as ".Constraints.Flange_Slope_Y * 6" which is reasonable and non-circular.  When loaded in recent builds, it gets removed.

The test in #11716 uses ".Placement" expressions and not ".Constraints" expressions, so it is certainly possible that is the change that triggers detection of this expression as potentially Circular; a test with ".Constraints" in the expression would clarify this point.

bgbsww, 2024-02-29:
Hmm:  [docs](https://wiki.freecad.org/Sketcher_ToggleDrivingConstraint) say "Reference dimensions" ... " can also be used to drive constraints in **another** sketch through expressions"

So the behavior of $11716 appears to match the documentation, but it causes a backwards compatibility issue in that a file that used reference dimensions in the same sketch in violation of the docs is now broken.

I'm not sure what the right next step is.

bgbsww, 2024-02-29:
Probably worth noting that using reference constraints in expressions in the same sketch could easily result in situations where the solver can't solve, which may well be where that documentation note originated.  It is a simplified way of avoiding potential circular references and infinite loops.  For example, make one line.  Now, make a reference constraint set to the length of that line.  Then make a driving constraint for the length of the line with a formula that includes the reference constraint.  I could be mis-thinking this somehow.

frank-trampe, 2024-02-29:
I am obviously biased because I have a few hundred hours in a project that I can't use in the old version due a bug in Techdraw and can't use in the new version due to this issue. With that disclaimer, I think that, even objectively, it makes sense to restore the previous behavior.

Being able to use reference constraints within the sketch is extremely handy. Visualizing a quantity rather than writing complex expressions based upon driving constraints makes the engineer's job easier and also decreases the chance of error.

With the utility of the feature, I think that there are probably a lot of other people who have used it. If this remains semi-silently broken, it could corrupt designs unexpectedly. I think that it is simpler to fix this than to handle the regression safely.

I do not know how the cycle-checking works, but I do know that there is a simple N^2 way to check within a sketch, and a sketch is unlikely to have enough constraints for that to be unacceptably slow.

bgbsww, 2024-02-29:
To your immediate problem, can you open, tweak the 6 self referential sketch references to use some other reference ( spreadsheet values, maybe? ), save and then use the newer versions?  That's clearly a sizable model, and you definitely need a way forward.

I did play with the sketcher with three lines ( one was too simplistic ), and there I can definitely make things get weird using reference constraints in formulas of the same sketch.

frank-trampe, 2024-02-29:
I could probably rewrite the formulae without much effort, but I feel like it might be a better use of my time to work on this bug rather than on the drawing so that I am stepping forwards rather than sideways. If I can figure out exactly when the breaking change landed, I can see whether the immediately preceding weekly build fixes my Techdraw problem.

I am able to make circular references in 0.21.2 with just two driving constraints. If I make them unstable (A set to B and B set to A * 2), evaluation stops after one iteration without a crash (with the two values doubling each time I edit anything). For all of the things that make FreeCAD crash, this is not one of them. 0.22.0dev.36082 correctly catches the cycle. The restriction on references to reference dimensions is clearly not a sufficient condition for avoiding cyclic dependencies, and it is not immediately clear (at least to me) that it is a necessary condition.

Could you perhaps test the current master branch with [the reference-banning logic](https://github.com/FreeCAD/FreeCAD/blob/97f9c4e467cba8e6d855abce2ca5499f3b7dc258/src/Mod/Sketcher/App/SketchObject.cpp#L8633) commented out and see whether cycle-checking of expressions dependent upon reference dimensions fails? I would do it myself, but I suspect that getting the build environment set up for such a big product as FreeCAD is likely to be non-trivial, and I want to avoid going down that hole unless absolutely necessary.

frank-trampe, 2024-02-29:
I was able to test this in FreeCAD 0.22.0dev.36082 without modifying any code. One can bypass the reference constraint restriction within the sketcher by setting it to drive and then setting it back to a reference constraint.

[Circular_Check_1_1_Archive_1.zip](https://github.com/FreeCAD/FreeCAD/files/14453766/Circular_Check_1_1_Archive_1.zip)

Open the file. Open the sketch. Set constraint Y to driving. Set constraint Y to reference. Then try setting the leftmost vertical line equal to the horizontal line constrained by X. FreeCAD somehow senses that something is wrong (either inconsistent/unbounded results or a cycle) and refuses. If you set X to a fixed value rather than one dependent upon Y, you can then set the equality constraint.

This demonstrates a robustness of the existing code against cycles or at least cycles with non-converging results even with references to reference dimensions within the same sketch. I would propose (subject to further testing by people with more direct knowledge than I have) simply removing the non-driving-constraint check.

frank-trampe, 2024-02-29:
Oh, wait. That's not consistently reproducible. I am going to figure out why I get different results.

frank-trampe, 2024-02-29:
Okay. Working around the reference restriction in FreeCAD 0.22.0dev.36082 results more often than not in the same behavior as in 0.21.2. The constraints recompute once per update. Rather than extending the cycle-checking functionality within the sketch, I think that the best option is to add a convergence check to the sketch solver and to warn the user if the solution is unstable or divergent. If this sounds like a good idea to others and nobody with experience with the relevant code wants to work on it, I will try to figure out a specific implementation.

bgbsww, 2024-02-29:
That's interesting.  I'm definitely not a specialist in the sketcher code or privy to any recent conversations on what direction it is headed in.  From a user standpoint, it makes sense to me, but hopefully someone more deeply versed will jump in.  I do think you want to watch out for situations that converge but result in different values each update pass leading to ever growing or shrinking values.

frank-trampe, 2024-02-29:
I was using converge to describe not just getting bounded output values (per the solver) but output values that are stable under recomputation. Given that other constraints do not have simple dependency relationships, cycle-checking might restrict a lot of valid and handy uses of reference constraints. I think that stable convergence is a more appropriate validation requirement and sufficient to avoid unexpected results.

frank-trampe, 2024-02-29:
To be clear, the stable convergence check could probably just take the form of caching all of the point positions, rerunning the solver, and checking that the values are the same as before.

frank-trampe, 2024-03-01:
I have tried a number of iterations of the build instructions and can't get FreeCAD master to compile. I patched out the reference constraint check in Sketcher.so in the previously mentioned weekly build, and it seems to work properly for me. Obviously not a permanent fix, but it keeps my project rolling. If we settle on a source-level fix for this, I will ask for help with compilation so that I can help with that.

maxwxyz, 2024-07-06:
Any news on this one?

kaiwas, 2024-09-17:
This is an important problem. I will really hope for its solution.

jonascant, 2025-03-01:
[flex-horn-muster-5.FCStd.gz](https://github.com/user-attachments/files/19038907/flex-horn-muster-5.FCStd.gz)
This File used to work fine to roughly sketch an exponential horn. This regression prevents it to recompute. Is there any workaround to disable this regression? Which older FreeCAD version still supports this?

closed 2025-06-29 by github-actions[bot]

frank-trampe, 2025-06-29:
@jonascant, the regression appeared after 0.21.2, so you can use 0.21.2. I also have a binary patched version of a weekly build for Linux/amd64 from February that bypasses the check and could share that if you like. (Same offer for @luzpaz and @kaiwas.) I am hoping at some point to get FreeCAD to compile in my environment so that I can patch at source level and add convergence checking.