FreeCAD/FreeCAD #12656: [Problem] Assembly : Using transform tool unset the active Assembly

PaddleStroke, 2024-02-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When you trigger the transform tool while editing an assembly, it will unset the edit of the assembly. Which is very confusing.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-28:
Will I be able to use the transform tool if the assembly stays active or will the transform tool *fight* with the joint constraints if I try to temporarily move the part aside?

luisDias89, 2024-02-28:
Why don't they perform a look-ahead on the constraints and make inaccessible the planes/rotations/directions that are not available in the transform tool?

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre