FreeCAD/FreeCAD #12657: Objects become transparent after save and reopen

marceyes, 2024-02-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A group of objects become suddenly transparent after save and open even after restart FreeCad.
Setting the colors of the individual objects her some copper clamps to standard is not permanent and will be forgotten after restart.
[image]
[BugObjectTransparent_20240228.zip](https://github.com/FreeCAD/FreeCAD/files/14438262/BugObjectTransparent_20240228.zip)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.4.56
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-28:
@marceyes could you please add steps how to reproduce it? If possible, please also try this with the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

marceyes, 2024-02-28:
OK:
Start FreeCAD.
Open File BugObject.... from zip container.
Select e.g. one copper clamp "TiegelHolderMasse" underneath "LinkGroupTiegel".
Right mouse button pull up menu and set color and "set to standard" and press OK.
Save file.
Restart FreeCAD and open BugObjectTransparent.FCStd file
and its again transparent.
Seems like something isn't entirely stored or I misunderstand the graphical rendering in FreeCAD.

xtemp09, 2024-02-28:
I cannot reproduce this bug on 

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

Maybe, it was fixed long time ago.

marceyes, 2024-02-28:
@xtemp09 I will try it again tomorrow after an update to V.21 and also on a different machine.
Thanks.

marceyes, 2024-02-29:
I tested it on a very old system with the identical FreeCAD-file and the transparency issue isn't appearing, however off course due to the old file format there are some disruptions in the collection of objects, e.g. wire is displaced, but ok.
Thus, it's most likely an OS problem with Windows 11 and drivers. I will further study this issue later on.

Just for completeness here the old setup info where it is working:
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

[image]

luzpaz, 2024-02-29:
@marceyes can you please test on dev build ([as requester earlier](https://github.com/FreeCAD/FreeCAD/issues/12657#issuecomment-1969547763)?)

Syres916, 2024-02-29:
@marceyes I don't believe this has anything whatsoever to do with drivers or even Windows. Quite simply there's an education issue with the way you change colours/transparency and there's an annoying bug in that the only way to get the correct colour/transparency to show is to set the whole document _Mark to Recompute_ after opening and then _Refresh_ which is completely against any 'normal' software principles that I'm aware of. It's highly likely this is as a result of you having a LinkGroup for the objects, something I would guess is not used by that majority of FreeCAD users on a daily basis but it should work so you were correct in raising this issue.

Education first, the _Set Color_ is only used for faces, if you need to change an object as a whole then you use _Appearance_ but there's also the _View_ tab in the _Properties Editor_ where you can tweak line colours, point sizes and transparency amongst other settings.

@luzpaz @maxwxyz the bug is still present with the latest main build, I even tested Realthunder's latest AppImage and it's only because his build asks the user if they want to do a Recompute on open that his correctly renders so IMHO there's something in the core code that's fundamentally not iterating the view object parameters correctly On Restore.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36218 (Git)
Build type: Release
Branch: main
Hash: a03ad7f49b1616439dae25a3c77539549cfc55de
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * InventorLoader 1.4.0
  * CurvedShapes 1.0.5
  * Design456 0.0.1
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * AirPlaneDesign 0.4.0
  * fasteners 0.5.10
  * sheetmetal 0.4.1
```

marceyes, 2024-02-29:
@luzpaz @maxwxyz @Syres916: Syres was right, it was an internal weakness of V0.20.1.
I tested it with the most recent version (FreeCAD_weekly-builds-36166-2024-02-26-conda-Windows-x86_64-py310) and this transparency issue seems to be solved now in this configuration:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36166 (Git)
Build type: Release
Branch: main
Hash: a5ea8023a29b6d7bbf1ebd57b5ed31dd27c5947b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.4.56
  * Help 1.0.3

In addition, if I open and save the BugObjectTransparent file then its size decreased by about 15%, indicating some significant improvements. But still, when loading the new file into the old version 0.20.1 (I know that's forbidden but anyway) then the copper clamps appear to be transparent. So I would take words of Syres seriously and look for some issues regarding the linked groups. May be, this transparency bug associated with the linked groups wasn't the only one.

I think, this bug is solved now and the Admin may close this thread at some time in future.
And for those who spent their time into it I have some little reward, an image how it looks like in reality. By the way, just to avoid other possible suggestions, this part is for basic research on solar energy conversion (in other words renewable energies) and not for commercial applications.

[image]

luzpaz, 2024-02-29:
@maxwxyz per @Syres916 

> @luzpaz @maxwxyz the bug is still present with the latest main build, I even tested Realthunder's latest AppImage and it's only because his build asks the user if they want to do a Recompute on open [...]

I'm going to re-open it, in that case.

maxwxyz, 2024-02-29:
I thought its not present in the latest build?

benj5378, 2024-10-04:
> OK: Start FreeCAD. Open File BugObject.... from zip container. Select e.g. one copper clamp "TiegelHolderMasse" underneath "LinkGroupTiegel". Right mouse button pull up menu and set color and "set [...]

I cannot reproduce this. There is no "Set to standard" in my pop up menu?

luzpaz, 2024-10-05:
cc @marceyes can you reproduce on RC2 ?

marceyes, 2024-10-05:
@benj5378 
With a newer Release obviously the Menu organization has changed. You have to keep the mouse pointer on Menu Item "TiegelHolderMasse" underneath "LinkGroupTiegel" while you activate the Pop-up Menu with the right mouse button and then "set color ..." and now one must actively select some or all surfaces (which is new) and "set to standard" and press OK. This procedure apparently overcomes the bug which I reported earlier and thus, with the actually installed Version 0.21.2 (rather than 0.19.2) on my old home notebook, the bug does not appear anymore.
As far as I remember previously one could select an entire object and make it transparent or non-transparent. Seemingly this function is now always broken down to the individual surfaces.
I can't say, whether the bug is entirely fixed since I'm not an expert in FreeCAD and I don't know the changes in the source code (which I would certainly be able to read given my experience with my charity project eyesage(dot)org). But it seems to me that this bug is not relevant anymore to most users. Thus, I would agree if @luzpaz would close this thread.
Thanks a lot to all people supporting this powerfull FreeCAD suit.

For completeness:
OS: Windows 7 build 7601
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.5.17

closed 2024-10-05 by luzpaz